MarlinFirmware/Marlin #12852: X and Y homing issues

geakg-gone-wild, 2019-01-07:
I get a "error: printer halted. (kill() called!" Any time I home my machine with X or Y axis. I have just installed a 3D touch auto bed level sensor, ironically it works fine. I was using Sprinter firmware but it dosnt support auto bed level, so thus im stuck using marlin. I used the latest version of marlin and as it did years ago and it gave me nothing but issues, printer wouldnt move or home at all. I used the marlin 1.1x bug fix addition and now every thing is working correctly except X and Y homing cause the machine to throw out the "error: printer halted. (kill() called!" on Repetier Host

When I use the M119 command all Endstops show as open unless triggered, the endstops are working correctly. When I home the X and or Y it will do as it should and stop when it hits the endstop but thats when the errors pop up and the printer freezes/ no longer to give commands with out disconnecting and reconnecting to the printer again.

If I home my Z axis it will deploy the probe and come down and touch the bed 2x and no error code and I can still manually move the printer around with the program movements after homing z axis. I'm not sure what would cause this to happen only on X and Y but any help would be appreciated I am at my whits end. 

I use repeiter host but i downloaded printerrun jsut to see what would happen and it had the exact same effect/ errors.
[upload files marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2734224/upload.files.marlin.zip)

geakg-gone-wild, 2019-01-08:
I tried the marlin bugfix 2.0 and it also has the same results.

shitcreek, 2019-01-08:
Please provide more information: what printer/board are you using?

geakg-gone-wild, 2019-01-08:
The Printer its self if a i3 MK3 Hybrid, I built my self.

the frame is made from this: https://www.thingiverse.com/thing:971110

all of the inner parts for X axis are using the Prusa i3 MK3 parts.

Y axis I designed my own parts for, but they work fine, I've already tested the printers movements using Sprinter firmware.

I should note I have A LOT of printer building experience, at least 20+ printers. I normally use sprinter firmware which I know really well since I use it a lot. I liked Marlins set up because it is a mostly the same as sprinter as far as how the configuration.h files work and most of its settings are identical. I would jsut use sprinter since it dosnt give me any issues but I'd really like to put auto bed level on all of my printers, this situation has me in a halt because I'd like to build another 10+ printers but I'd really really like for them to have auto bed level.

GMagician, 2019-01-08:
You may try enabling ENDSTOP_NOISE_FILTER. If it solves you have electric noise on endstop inputs

geakg-gone-wild, 2019-01-08:
I did try that already, it didnt change any behavior. I am using optical endstops though, some thing I forgot to mention.

GMagician, 2019-01-08:
Do printer halt when endstop is "pressed" or after motor move a little out of sensor?

GMagician, 2019-01-08:
If pull up solve homing it means sensor don't have it and just close to GND.
Do you have M120/M121 in your gcode?

geakg-gone-wild, 2019-01-08:
I havnt added any G-code so I dont think I have the M120/M121 in my G-code.

Also, its only the Y axis causing the issue now. When I try to print X axis homes and then Y goes to home soon as it touches senor it gives me the error.

GMagician, 2019-01-08:
> When I turn on pull ups for X and Y min I can now home the printer

you can home axes...

> When I try to print X axis homes and then Y goes to home soon as it touches senor it gives me the error.

you can't home when you start printing..

do I understood correctly?

GMagician, 2019-01-08:
...maybe pull up is nopt strong enough

geakg-gone-wild, 2019-01-08:
yes, correct whn i try to print errors come back. It seems some times when I home it gives the error randomly now, some times it homes fine some times it gives error.

I think it is firmware issue because when I install sprinter firmware every thing works fine with no issues at all but then I cant use auto bed level. with sprinter installed instead of marlin the printer works perfectly but sprinter does not have support for auto bed level feature.

Micro-3D-GER, 2019-01-13:
**@Hello georg-gone-wild**

What endstops did you build for X and Y?
I have built these mechanical endstops for X and Y, here is an example [https://www.ebay.de/itm/3-Stucke-Mech-Endstop-Schalter-Reprap-Makerbot-Prusa-Mendel-RAMPS1-4-Cnc-3D-ne/332657061391?hash=item4d73e7ee0f:g:AG4AAOSw5KFbPdzh](url)

This is how I configure the X and Y MIN endstops
```// Enable pullup for all endstops to prevent a floating state
#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  #define ENDSTOPPULLUP_ZMIN
  #define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the probe.```

boelle, 2019-02-21:
@geakg-gone-wild problem solved?

closed 2019-03-03 by thinkyhead