MarlinFirmware/Marlin #12860: EEprom version mismatch, how to fix?

boelle, 2019-01-09:
How is it that we change the eeprom version number?

i did the usual M502 and M500 then M501, but the mismatch in version number is still there

i think its because i tried the a copy of marlin 4 days ago but it has a "bug" that prevents me from controlling a 2nd part cooling fan, so i had to go back to a copy from 5/11 that works

i read that people with radds/due used to flash repetier and then back to marlin to solve it, but that is not an option here, plus the printer is in another room and onboard sd-card of the REARM is hard to get to as wires pass close by, i can do it but i want to limit the amount of walking back and forth and taking the sd card in and out

RudyBenner, 2019-01-09:
This may work. had the problem a week ago. Since I cannot recreate the problem, I cannot say this is a solution.
Compile with //#define EEPROM_SETTINGS
Then compile without forward slash like this #define EEPROM_SETTINGS.
It could be that I just got lucky.

gloomyandy, 2019-01-09:
I think that normally M502 and M500 then M501 would work, but you may have hit problems that existed in the eeprom emualtion code in that older version that means it does not update things correctly (There was a bug with memory alignment in the LPC1768 code), that bug has since been fixed. You may be able to fix it by downloading a current version of persistent_store_flash.cpp:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_LPC1768/persistent_store_flash.cpp 
and replacing that file in your 5/11 build.

boelle, 2019-01-10:
will give it a go later in the day, but i that much that the problem is only due to that eeprom now have a newer version number than the firmware does

boelle, 2019-01-10:
changing the version number just throws a crc error

but the version error was gone so i tried to hit init eeprom from the display as i did no not know the code to send over serial. will try and change version back and do an init from display again

gloomyandy, 2019-01-10:
As I said above M502 and M500 then M501 is normally enough to clear the eeprom and write a new version of everything including the version number. But the old version of the code that you are trying to use has a bug that means that nothing will be written to eeprom in some circumstances (the buffer used to store the eeprom data is not correctly aligned) which means that the M502 and M500 then M501 will not clear and fix the version number. You just end up using whatever is already in there. To fix the version number you need to have code that can write to eeprom and at the moment you don't have that.

I don't think messing around with the version number is going to help you fix the basic problem you have. It may look like it has (because you may get lucky and change the memory alignment of things so that the write works), but the problem will likely as not come back if you change anything else.

boelle, 2019-01-10:
i would have just upgraded to the latest copy but with that i loose control of the extra large bed cooling fan i use after print is done, i have a seperate issue on that but with my limited skills i cant figure that even thou i try answering the best i can when others reply to that issue: its this issue https://github.com/MarlinFirmware/Marlin/issues/12831

boelle, 2019-01-10:
just replaced persistent_store_flash.cpp and building, will report back later

boelle, 2019-01-10:
worked fine

so think i will close this one again unless it can serve a usefull purpose like being the basis for a feature or something like it

closed 2019-01-14 by boelle

HaroldBawls, 2019-10-16:
I got this issue when switching from Marlin 1.1.9 to Marlin 2.0.x

I fixed this issue by following instructions from issue 11415. Setting **ALL** EEPROM values to 0 and then flashing the new firmware to the board fixes it.

Roxy-3D, 2019-10-16:
Doing a M502 followed by a M500 should resolve any EEPROM version mismatch...

6XoCtujg2C0gne, 2020-02-29:
I'll add to what @Roxy-3D commented and add a reboot.
I just encountered this with 2.0.4.4 and after the reboot all clear!

M502 > M500 > board restart

cnyper, 2020-03-02:
> I'll add to what @Roxy-3D commented and add a reboot.
> I just encountered this with 2.0.4.4 on my LPC1768 and after the reboot all clear!
> 
> M502 > M500 > board restart

Same here... Ender 5,  2.0.3 -->2.0.4.4 'ERR: EEPROM Version' showed at bottom of start up screen
M502 M500, Reboot got rid of it...
had to remove PIDBed as 2.0.4.4 upload was 172bytes over... (Note to self... get 32bit board!)

InsanityAutomation, 2020-03-02:
FYI, There is an EEPROM_AUTO_INIT option to do it for you when needed...

cnyper, 2020-03-02:
> FYI, There is an EEPROM_AUTO_INIT option to do it for you when needed...

I didn't have it implemented in 2.0.3..
I was going to ask... "How much space would it take up if I enable that setting..." as I wasn't able to add PID_Bed without going over... But I thought I'd better not be lazy and to check myself...

Seems by uncommenting, I still have 800 bytes spare... let's see what happens when I upload!

A

aurion55, 2020-03-15:
Help please, I cannot connect 3D printer via Pronterface after upgrading to 2.05, I get this error: Attempted to write invalid text to console, which could be due to an invalid baudrate
So reflash with 2.05 to No avail?