MarlinFirmware/Marlin #12861: Power loss recovery resume position error

fatihbircan, 2019-01-09:
Hi, I am new in Marlin and have question for you. I am sorry for my bad english. I tried to use power loss recovery. It worked good but there is only one problem. When I clicked resume: nozzle first going home position and after that it is trying to go last position(when power off). But the problem is that it's never resume from exact last point. Also resuming point difference is not negligible because it is at least 4-5cm. You can think like that I am trying to draw a line which is from point 0 to point 15. When it reached the point 5 power off. Power on: When I clicked resume it is going point 10(not 5) and resume. I have no problem with Z axis, I fixed it.

AnHardt, 2019-01-09:
Some of the developers are well aware of this problem and for that never implemented resume from power loss. To make 'resume from power loss' happen, at all, this problem was simply ignored.
Currently, when a power loss is detected all movement is stopped immediately and the position in the g-code-file is stored. That means, all contend, currently in the planner buffer, is lost. 
An alternate strategy, where the 'stop' does not happen immediately but after the content of the buffer is printed, does not work, because this could last several seconds - too long for the remaining power.
The other alternative is stopping the movement immediately, but saving (and later restoring) the planner buffer. Sadly this is nor a weekend project. I sketched a concept [here](https://github.com/AnHardt/Marlin/issues/87) for an other connected variante of the problem.

boelle, 2019-02-21:
@fatihbircan problem solved?

closed 2019-03-03 by thinkyhead