MarlinFirmware/Marlin #12862: Customization of Firmware

TigerClaw33, 2019-01-09:
Hello, 
I want to build my own printer and have a question about the cutomization of firmware. Could I use it as standard for every printer or is it necessary to customize it for every printer?

How is it possible to define the distance between the stepper? How do they know how to drive for example only inside a specific area? First I have to set my NULL point. Or is the endpoint defined by end-stoper? If yes how do I know how many steps I have to drive in the other directions?

Sorry for this basic question.

swilkens, 2019-01-09:
I would suggest that you begin by reading here:

http://marlinfw.org/docs/configuration/configuration.html

It is very possible and even required to configure the base firmware for the hardware you are using. Many hardware options have been pre-configured and are user selectable at compile time. This includes stepper, stepper driver, steps per mm, end-stops, control board and many other features.

After reading the documentation, further questions about **using** and **configuring** Marlin for your printer should be asked on the official facebook group for the Marlin firmware:

https://www.facebook.com/groups/1049718498464482

If, after that, you still have problems with some features, your configuration or compiling the firmware - this github is the best location to ask about **bugs** and **issues**.

Finally, if you want to help yourself, try to select a control board which is supported by the firmware. The many variations of the RAMPS shields for Arduino boards is a solid classic base.

I hope this helps as a starting point, good luck with your printer build :)

swilkens, 2019-01-11:
If this has answered your question, please remember to close this issue report. A button exists on this page, look for the "comment and close" button 👍

boelle, 2019-02-21:
@TigerClaw33 problem solved?

closed 2019-03-03 by thinkyhead