MarlinFirmware/Marlin #12864: marlin, bltouch 1.1.9 & 1.1.8

MrDirt, 2019-01-09:
C10s same error

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

GMagician, 2019-01-09:
Please fill form correctly, I don't really understand what's your problem

MrDirt, 2019-01-09:
I apologize this is the first post I made  I thought I did it correctly

 
PRINTER CREALITY10S
Configured marlin 1.1.8 for added hardware BL Touch could not compile the sketch\
I expected the sketch to compile 

Actual behavior:
I received the following error both times

> static assertion failed: Z_SAFE_HOMING_X_POINT is outside the probe region.

 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2742888/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2742894/Configuration_adv.zip)

RudyBenner, 2019-01-09:
//#define BED_CENTER_AT_0_0

Easy fix.

MrDirt, 2019-01-09:
Thank You Sir!

closed 2019-01-09 by MrDirt