MarlinFirmware/Marlin #12865: marlin2.0 / sbase / external driver

dalgomsha, 2019-01-10:
Hello.
I'm making a big 3D printer, but I'm in trouble.
I ask for your help.

Initially manufactured using
ramps1.4 + arduino2560 + 12864 full graphic display (12v)
dm542 dsp motor driver (42v)
600 mm actuator x5 (nema24 motor / x1 y2 z2)
x,y,z motor steps_per_mm is 133.33.

Simple modeling worked well, 
but complicated things like the Eiffel Tower were made diagonally in the Z-direction.
i don't think it was an layer shift.

So I decided to buy a mks sbase board and use the Marlin 2.0 firmware 
because I thought arduino 2560 is not enough computing power.

The firmware upload on the sbase board was well done and 
it was verified to work well using the built-in drv8825.

So I used the external motor driver pin to apply the dm542 dsp motor driver, 
but the motor does not work.
auto home, move axis is doesn't work.

I studied the firmware very hard.
external motor driver Pin is active and no part of external motor activation was found.

I tried to connect in two ways here, but it didn't work.
http://www.smoothieware.org/3D-printer-guide#external-stepper-drivers

I'd appreciate it if you could tell me what I'm missing.
thank you.

FanDjango, 2019-01-10:
What wiring have you chosen? Common Anode or Common Cathode?  Because the DM542 would prefer common Anode - which means programming the pins to be open-drain. Dunno if Marlin can do this.

If you wire as Common Cathode (ENA-, DIR-, PLS- to ground, ENA+, DIR+ and PLS+ to the MKS controller board, which only supplies 3.3V to the DM542 input circuit, it won't fire the optocoupler input. You would need a level shifter to 5V signal.

I referred to this link: [https://cdn-reichelt.de/documents/datenblatt/A300/DM542-DEUTSCH.pdf](url) as well as your link to smoothieware.

boelle, 2019-02-21:
@dalgomsha problem solved?

dalgomsha, 2019-02-22:
> @dalgomsha problem solved?

Hey, buddy.

I have tried it several times, but I have come to the conclusion that 
the dm542 cannot be a marlin2.0 firmware.

http://www.smoothieware.org/3D-printer-guide#external-stepper-drivers

dm542 dsp motor driver does not support the link 
'Wiring an external driver with a common cathode'
Only supporting an 'Wiring an external driver with a common anode' 

Marlin2.0 is only available with 'Wiring an external driver with a common cathode'

I tested it for one MA860H motor driver.
It works well with the marlin2.0 firmware.

So I tried to change all the motor drivers to MA860H. 
In the end, Marlin 2.0 gave up and decided to use Smoothieware.

Now it works well with smoothieware.

boelle, 2019-02-22:
oki click close please

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-11 by thinkyhead