MarlinFirmware/Marlin #12866: ADVANCED_PAUSE_FEATURE currently requires an LCD controller error

rouhit110, 2019-01-10:
Hi friends
I have a new problem with my firmware
I recently installed a filament run out sensor and I made the settings that were needed

Configuration.h
#define FILAMENT_RUNOUT_SENSOR
Configuration_adv.h
#define ADVANCED_PAUSE_FEATURE

I use MKS tft 2.8 V2 display and MKS Gen 1.4 and Arduino 1.6.13

#error "ADVANCED_PAUSE_FEATURE currently requires an LCD controller."

http://s9.picofile.com/d/8348489734/0f28a0a4-41f0-4766-8993-22e55104dd34/Configuration.h Configuration.h
http://s8.picofile.com/d/8348489784/2f44c1a0-34ac-4c48-8691-43519b3be9c8/Configuration_adv.h Configuration_adv

Please see this files attached and help me dears
Thanks in advance for your help...

FanDjango, 2019-01-10:
**#define CONFIGURATION_H_VERSION 010107** Yeeks, which Marlin was that? 

First thing I would do, I suppose, is upgrade to a current Marlin version. Any research into whether this has been fixed by now is time consuming maybe.

FanDjango, 2019-01-10:
And it was fun clicking on some of those:

[image]

rouhit110, 2019-01-10:
Hi dear @FanDjango . please click on this item
[image]

I use marlin 1.6.13

FanDjango, 2019-01-10:
arduino 1.6.13 is very old. We are on 1.8.8 nowadays. But the real issue is your Marlin. CONFIGURATION_H_VERSION 010107. Maybe more than a year old.
If you stick with 1.1.x, we are at 010109 right now. Think about upgrading first before anyone researches this issue.
Also check this issue, for example: #7466. Or #6982, so there was a lot of stuff concerning symptoms of this kind.

FanDjango, 2019-01-10:
Ok, but now I am confused.
If your **configuration.h** contains **#define CONFIGURATION_H_VERSION 010107** you should get an error message when you compile it because the configuration file version 010107 does not correspond to the Marlin 1.1.9 (which is the current 1.1.x from [http://marlinfw.org/meta/download/](url)). So something is wrong.

FanDjango, 2019-01-10:
When you download the newest Marlin, look at the configuration.h that comes with it. You will see the **#define CONFIGURATION_H_VERSION 010109**. You cannot simply take your old configuration.h (the one from the old 1.1.7 version (010107) and use it. You must place the old one and the new one side by side so to speak, and update the new one with the old values you need, without destroying the new parameters that may have been introduced with the new versio,
I use a compare-tool like ExamDiff Pro (Windows) for that, or on linux use diff or other compare tools.

rouhit110, 2019-01-10:
Thank you dear @FanDjango for giving me valuable time and guiding me.
My first problem with the marlin update was fixed and I no longer encountered that error
As you said, I did all the values manually
Just do the final test tomorrow and I'll be sure to let you know

thinkyhead, 2019-01-10:
For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

closed 2019-01-10 by thinkyhead