MarlinFirmware/Marlin #12867: Bug with Filament Runout Sensor Script/ Advanced Pause Features

forum999, 2019-01-10:
despite being enabled marlin 1.1.9 seems to have a bug with the filament runout sensor script.

When i try to compile i get the following error despite ADVANCED_PAUSE_FEATURE being uncommented and setup (see attached pic).
[image]

`In file included from sketch\MarlinConfig.h:42:0,

                 from sketch\G26_Mesh_Validation_Tool.cpp:27:

SanityCheck.h:502: error: static assertion failed: ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.

     static_assert(NULL == strstr(FILAMENT_RUNOUT_SCRIPT, "M600"), "ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.");

     ^

exit status 1
static assertion failed: ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.`

FanDjango, 2019-01-10:
Post the complete configure.h and configure_adv.h, maybe?

forum999, 2019-01-10:
Apologies, i thought i had added them.

[config.h and config adv.h.zip](https://github.com/MarlinFirmware/Marlin/files/2746240/config.h.and.config.adv.h.zip)

forum999, 2019-01-10:
I believe its the one at the top where i clearly state this is from Marlin 1.1.9? Lol

I wanted to update my current version of marlin so i didnt have to add all the changes etc into 2.0

FanDjango, 2019-01-10:
Hmmmm, that's funny, because your configuration.h contains a
```
#define CONFIGURATION_H_VERSION 020000
```
which led me to believe you want 2.0.x.

You might want to recheck the whole setup.

FanDjango, 2019-01-10:
I should add, having a higher version in there than "needed" passes the sanity check with no errors being raised. I haven't checked the rest of the file as to whether it conforms more to 1.1.9 or more to 2.0.0, that's not the important point. **You are right, it probably got fixed sometime between 1.1.9 (yours) and Bugfix-2.0.0 (current git version)**

forum999, 2019-01-10:
Hmm that is odd? Definately 1.1.9 though?

boelle, 2019-02-21:
@forum999 problem solved?

closed 2019-03-03 by thinkyhead