MarlinFirmware/Marlin #12870: [32-bit DUE] Extruder no longer steps when Linear Advance is enabled.

marcio-ao, 2019-01-10:
I've been trying to track down a weird bug in our 32-bit firmware. In at least one variant of our FW, the extruder no longer steps when LIN_ADVANCE is enabled (with K = 0), but steps when LIN_ADVANCE is disabled.

To try to narrow down where bug might be, I commented out all the LIN_ADVANCE stuff in "planner.cpp" and verified that still no extrusion was taking place. I am led to believe that the bug is hiding in "stepper.cpp". There are a lot of places in "stepper.cpp" where stuff gets conditionally compiled based on LIN_ADVANCE, but it appears to all be inter-related, so I don't think I could just comment some bits and not others to try to narrow this down any further.

Any ideas on what else I could try?

Xilius, 2019-01-16:
I can second it. Using SKRV1.1 and the extruder stepper only moves when LIN_ADVANCE is disabled

gloomyandy, 2019-01-17:
@marcio-ao What happens if you have a non zero value for k? I've just run a test with a recent build (couple of days old) and things seem fine for me. I have LIN_ADVANCE enabled and in my case k was 0.1. Print worked fine. I'm running on an SBase LPC1768 board with a pretty vanilla Cartesian setup. Let me know if you think trying with K=0 would be of any help. Oh and I have S curve and adaptive stepping enabled.

marcio-ao, 2019-01-17:
@gloomyandy: I didn't test it with K != 0. We want to enable LIN_ADVANCE for our customers who want it, but since we have not validated our print profiles for LIN_ADVANCE, we keep the K value at zero by default. It's entirely possible the bug only crops up with K = 0, as this might be a corner case.

Xilius, 2019-01-17:
The bug is still persistant with K!=0, K=0.1 , default 0.22 or even k=10 it doesn't matter.

gloomyandy, 2019-01-17:
I've just tried with k=0 and that worked fine as well for me, so whatever it is it is not as simple as just having liner advance enabled. 

Does the extruder make any attempt to move? Any odd sounds? Does reducing the feedrate make any difference? What happens if you remove the filament (to reduce the motor loading) will it extrude then? Oh and I have a geared extruder (Titan Aero) so if there is a loading issue, i will probably have less.

marcio-ao, 2019-01-17:
@gloomyandy: We compile firmware for various machines and toolheads, and the problem only shows up on certain builds. So you are correct that it's not as simple as just having linear advance enabled. I have not, however, been able to identify the exact combination of features that causes it. I only know that disabling LIN_ADVANCE solves it. We might be dealing with a schrödinbug here...

marcio-ao, 2019-01-17:
Incidentally, we are using Titan Aeros as well. There is no extrusion and no noise. In fact, if we hook up an o-scope to the driver lines, we see the PWM signal remains steady, so we know no stepping is actually taking place.

gloomyandy, 2019-01-17:
Oh dear, that sounds like it is going to be hard to identify what is going on. Will it ever extrude? So if you just do a movement on E (with no move on X/Y) does that work?

marcio-ao, 2019-01-17:
@gloomyandy: No, the extruder is dead. Dead as a doornail.

marcio-ao, 2019-01-17:
@gloomyandy: I know for my experimentation that steps are getting correctly inserted by the planner into the blocks, so it's really something in the ISR. We've had other issues crop up before due to differences between how the ISRs are handled on Due vs. AVR, so if I were to make a guess as to what is going on, I would suspect it has to do with the fact that LIN_ADVANCE seems to run it's own ISR (as far as I can tell).

gloomyandy, 2019-01-17:
Hmm unfortunately I think all you can do (unless someone that knows that code better than probably both of us turns up) is to start adding trace code into that stepper code and see if it is doing what you would expect. Not easy I know, but the good news is that it seems to be a hard failure and does not come and go. Good luck!

marcio-ao, 2019-01-17:
@gloomyandy: I was able to trace it up to the planner code, but once I got into the stepper code it was out of my league...

Xilius, 2019-01-18:
> I've just tried with k=0 and that worked fine as well for me, so whatever it is it is not as simple as just having liner advance enabled.
>
> Does the extruder make any attempt to move? Any odd sounds? [...]

hooked up a spare stepper motor without any load - it intends to move a tinytiny bit and the motor is locked but doesn'T continue - changing feedrate acceleration and so on has no effect

kookaburra88, 2019-02-10:
SKRV1.1 --> Same issue.

gloomyandy, 2019-02-10:
Just in case anyone reading this thread has not realised the following issue contains what seems like a work around, so it is probably best to try this and comment over there (especially if it does not fix things for you).
https://github.com/MarlinFirmware/Marlin/issues/13040

marcio-ao, 2019-02-18:
Closing this ticket in favor of #13040

closed 2019-02-18 by marcio-ao