MarlinFirmware/Marlin #12871: Ender 3 Configuration is using the wrong jerk setting

Mnemonic7, 2019-01-10:
### Description
Hi, the configuration.h file for the ender 3 has the jerk value set to : 
#define DEFAULT_XJERK                 10.0
#define DEFAULT_YJERK                 10.0
#define DEFAULT_ZJERK                  0.3
#define DEFAULT_EJERK                  5.0

end the original ender 3 configuration file uses this settings:
#define DEFAULT_XYJERK                20.0    // (mm/sec)
#define DEFAULT_ZJERK                 0.4     // (mm/sec)
#define DEFAULT_EJERK                 5.0    // (mm/sec)

So the configuration.h for ender 3 should be:
#define DEFAULT_XJERK                 20.0
#define DEFAULT_YJERK                 20.0
#define DEFAULT_ZJERK                  0.4
#define DEFAULT_EJERK                  5.0

I did this change and didn't notice any change in print quality so we should keep with the original values created by creality.

thisiskeithb, 2019-01-11:
When I created the request, it was my understanding that the jerk settings were halved at one point during Marlin development. Creality was using a version of Marlin from before this change, so I adjusted the settings appropriately.

swilkens, 2019-01-11:
Creality recently, upon request, made a more up to date version of their firmware source available:

https://www.creality3d.cn/download/firmware_c0001

That link now provides newer sources for the Ender 3 and Ender 3 Pro, now based off what seems to be Marlin 1.1.6. The configuration files included there show:

> /**
>  * Default Jerk (mm/s)
>  * Override with M205 X Y Z E
>  *
>  * "Jerk" specifies the minimum speed change that requires acceleration.
>  * When changing speed and direction, if the difference is less than the
>  * value set here, it may happen instantaneously.
>  */
> #define DEFAULT_XJERK                 10.0
> #define DEFAULT_YJERK                 10.0
> #define DEFAULT_ZJERK                  0.4
> #define DEFAULT_EJERK                  5.0

reloxx13, 2019-01-11:
after those jerk changes in the config i got heavy y layer shifts. im going back now and see if this was the reason.

closed 2019-01-14 by thinkyhead