MarlinFirmware/Marlin #12873: Infill 100% Problem _ [Anet A8 - Bowden - Marlin 1.1.9]

Lucaevo2, 2019-01-11:
Hi guys
I have a problem with my anet.
Before switching to the Bowden and the Marlin 1.1.9 firmware, the printer printed fine.

Now I have this problem of infill. It almost seems like a steps issue.

In fact I tried to change some parameter in the code:
---------------------------------
**1)**
 _#define DEFAULT_AXIS_STEPS_PER_UNIT {**80, 80**, 400, 100}_

Since:
Step Angle 1.8 °
Pulley Tooth 20
GT Pulley 2mm

With this I solved the problem of the infill m m the size of the calibration cube were wrong.

-------------------------------------
**2**
_#define DEFAULT_ACCELERATION **250**             //X, Y, Z and and ACCELERATION for printing moves_

_#define DEFAULT_XJERK **3.0**
#define DEFAULT_YJERK **5.0**_

The result is improved but not solved.
-----------------------------------
**3)**
 I have also returned to its original firmware. By printing better but the problem is not fixed.
----------------------------------

I would not go back to direct extrusion, but I would solve this problem.

Does anyone have any ideas or better, a solution?

[image]

[image]

vilsed, 2019-01-11:
It's not clear - are the external dimensions of your test cubes incorrect or are you just not satisfied with incomplete solid infill filling? Because if it's the latter - all you have to do is calibrate your E-steps/mm.

Lucaevo2, 2019-01-11:
Sorry about my explanation. My problem concerns precisely with incomplete filling of the solid fill.
Now I try to calibrate my E-steps/mm.
Thank you so much for the support.
I'll keep you updated.

Lucaevo2, 2019-01-14:
I calibrated the e-steps, but I did not solve it. I think it's a problem of setting up firmware when printing fills. I have the same problem:

https://youtu.be/Icw-mR_mzhg

vilsed, 2019-01-15:
Interesting. Firmware has no business or, for that matter, no idea "what" it's printing. It does not matter if it's 20% infill, 100% infill or perimeters, firmware does not care. It just executes gcode commands it is given. Please share your configuration.h, configuration_adv.h and gcode file you're trying to print.

Lucaevo2, 2019-01-15:
thank you for your support. 
I share the files:

configuration.h
configuration_adv.h
.gcode
.factory

now, I'm using Marlin 1.1.9 and simplify3d

[files.zip](https://github.com/MarlinFirmware/Marlin/files/2759301/files.zip)

Lucaevo2, 2019-01-15:
I don't know if this can help.

On the web I found K-factor Calibration Pattern.
link: http://marlinfw.org/tools/lin_advance/k-factor.html

I also share the g.code generated by the website with the stock setting and the result of the print

[K-factor.zip](https://github.com/MarlinFirmware/Marlin/files/2759365/K-factor.zip)

vilsed, 2019-01-15:
Very very interesting. I didn't saw anything out of the ordinary in your files. Did you calibrate the extruder simply by extruding set amount and measuring how much filament was actually extruded? I find this method good only for rudimentary calibration, to get "close enough". For precise calibration I print with 100% infill and increase/decrease the E_steps/mm with M92 until I just achieve actually solid infill without gaps and without over extrusion. One more thing to check - does your filament move freely through the bowden tube? Is the extruder not skipping steps? If all those check out - try printing a bit slower, maybe at 30mm/s. If everything fails - you could try bugfix-1.1.x, but I still highly doubt that this is firmware related. I've had similar problems before, all times it was hardware/calibration related.

vilsed, 2019-01-16:
As far as extrusion  rate goes - I think you have it pretty much dialed-in for a bowden setup. As for the gaps and size mismatch in Y direction, I think it's a mechanical issue with the bed. Make sure that it can move freely. Make sure that the pulley on the motor is tight and not wobbly. If you have spring tensioners on your belts - remove them. They are more trouble than they're worth. These cheap chinese machines are notorious for problems like these. 

If everything seems correct, try lowering jerk and acceleration for the Y axis. 

One other problem I see with your test cube - lack of cooling/overheating. I'd suggest to always keep the part cooling fan at >80% for PLA. But this is completely unrelated.

Lucaevo2, 2019-01-17:
I followed your suggestions and found that it was a mechanical problem of the bed. I disassembled all the axes bars and I cleaned them and oiled them.
Now it's fine!

The size of the y was wrong because of the friction, like every filling problem!

Now it's perfect

Thank you very much!

vilsed, 2019-01-18:
Glad to help :)

boelle, 2019-02-21:
@Lucaevo2 problem solved?

closed 2019-03-03 by thinkyhead