MarlinFirmware/Marlin #12875: Einsy Rambo Fan causes freeze

Lyr3x, 2019-01-11:
Since today i have a weird problem with my Einsy Rambo. This mainboard has two controllable fans and as soon as layer FAN 1kicks in the motors wont move anymore until a reset. Heaters/Thermistors still working and i also can use the LCD. 

When connecting the Fan to the FAN 2 connector and start the fan everything is working fine. I tried to change the Pin mapping to FAN 1 is now the FAN 2 connector. After firmware upload and eeprom reset the behavior was the same. The fan is connected to the FAN 2 connector but declared as FAN 1 and when the fan start all movement stopped. 

Any Ideas?
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2750387/Config.zip)

revilor, 2019-01-11:
I don't know if this is relevant for your issue, but in Prusa Firmware the fan is on PIN 6

#define FAN_PIN              6
#define FAN_1_PIN           -1

AnHardt, 2019-01-11:
Try `#define FAN_SOFT_PWM` looks as if the hardware timer used for that fan-pin-hardware-pwm  is already in use for the stepper interrupt.

Lyr3x, 2019-01-14:
I'll have a look at this next weekend and report to you 👍

boelle, 2019-02-21:
@Lyr3x problem solved?

closed 2019-03-03 by thinkyhead