MarlinFirmware/Marlin #12877: [bugfix 2.0.x] LPC1768 Compile Error

556duckvader, 2019-01-11:
While trying to compile the latest 2.0.x version for my Re-ARM I am getting a compile error.
!
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1768; framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x > NXP LPC1768
HARDWARE: LPC1768 100MHz 31.80KB RAM (464KB Flash)

DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(strict)
Collected 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.2.3
CalledProcessError: Command 'fsutil fsinfo drives' returned non-zero exit status 1:
File "C:\Users\dbnorton\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 175:
env.SConscript(item, exports="env")
File "C:\Users\dbnorton\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\dbnorton\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\dbnorton\Desktop\Marlin-32bit\NewVersion\Marlin-bugfix-2.0.x\Marlin\src\HAL\HAL_LPC1768\upload_extra_script.py", line 36:
driveStr = subprocess.check_output("fsutil fsinfo drives")  # typical result (string): 'Drives: C:\ D:\ E:\ F:\ G:\ H:\ I:\ J:\ K:\ L:\ M:\ Y:\ Z:\'
File "C:\Python27\Lib\subprocess.py", line 223:
raise CalledProcessError(retcode, cmd, output=output)
 [ERROR] Took 1.69 seconds
Environment LPC1768                     [ERROR]
 [ERROR] Took 1.70 seconds
 
 [SUMMARY]
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment at90usb1286_cdc             [SKIP]
Environment at90usb1286_dfu             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1769                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [SKIP]
Environment ARMED                       [SKIP]
Environment mks_robin                   [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
Environment fysetc_f6_13                [SKIP]

p3p, 2019-01-11:
What OS is this? (Windows 7?) I'm not that familiar with Windows but it looks like a permissions issue to me, fsutil may need "run as Administrator", 

Just removing the upload drive autodetect script `upload_extra_script` from `platformio.ini` will probably fix it, the script needs more error checking.

556duckvader, 2019-01-11:
Thank you very much that fixed my issue.

closed 2019-01-11 by 556duckvader