MarlinFirmware/Marlin #12880: issue with homing on Anycubic Kossel, stops before hitting the endstops

KA8888, 2019-01-11:
Hello,
this week i updated my Anycubic Kossel with the newest 1.1.9 and then 2.0 bugfix.
Since then its calibrating the height much shorter then before. The height was at around 335mm and now only 318mm. The difference is the z-probe distance from the nozzle.
Although everything seems to be right and the nozzle touches the bed at the right height if I let it down to z0. But as soon as I home the printer while the nozzle is that low, its showing an error. It happens on both versions 1.1.9 / 2.0 and even if i deactivate the software endstops.
Any idea whats causing this?
Or is that maybe a bug?

[configfiles.zip](https://github.com/MarlinFirmware/Marlin/files/2751449/configfiles.zip)

TheMrCaveman, 2019-01-12:
Do you remember the z-offset? If you remember set it before you do anything else. Had the same issue and this fixed it for me. The whole readme is not accurate anymore after the G33 fixes and menu changes.

GMagician, 2019-01-12:
M502 and then M500 to update eeprom content...if you have it

KA8888, 2019-01-12:
Thank you for the replies.

@GMagician  
I did. Before I update I additionally use a sketch to clear the eeprom.
`#include <EEPROM.h>

void setup() {
  Serial.begin (9600);

  for (int i = 0 ; i < EEPROM.length() ; i++) {
    EEPROM.write(i, 0);
  }
  delay (100);
  Serial.println ("eeprom cleared!");
}

void loop() {
}`
@TheMrCaveman 
Im not sure what you mean with "set it before you do anything else". I remember the z-offset, its -16,19 and I set it in the configuration.h. After that I update my printer, clear the eeprom and start the delta autocalibration.

GMagician, 2019-01-12:
> as I home the printer while the nozzle is that low, its showing an error.

which error?

KA8888, 2019-01-12:
The screen is saying that homing failed, I should restart the printer.
Testet it with a command prompt as well and getting this message: Error:Printer halted. kill() called!

GMagician, 2019-01-12:
Ok your Z is near 0 you ask for homing and...motor rise or error is immediate?
Sure you don't have noise on endstops? 1.1.9 and 2.0.0 changed some behaviours on endstops and now are more sensible to noise
Try to enable ENDSTOP_NOISE_THRESHOLD

KA8888, 2019-01-12:
I think its about the delta hight. If I lift the nozzle/effector to the highest point, I messure over 330mm from the nozzle to the bed. But the delta autocalibartion sets the hight to 318mm, its subtracting the z-offset of the probe. Earlier builds didnt do this. Thats why my old Firmware (1.1.6) had a delta hight of around 335mm. I think its somethink about the safety features, basically the nozzle/effector is traveling on the z axis higher than the delta hight.

TheMrCaveman, 2019-01-12:
With the latest 2.0 branch i have the same issue. I fixed it by manually setting the z-offset after clear eprom. Then usual calibration stuff goes fine and homing works properly.
Lately the z-offset bug is fixed and set delta height is removed from menu. But the included readme is never updated so that is very confusing!

KA8888, 2019-01-13:
I tried it with ENDSTOP_NOISE_THRESHOLD enabled, as expected there is no change.

@TheMrCaveman 
What do you mean?
Shoud i let the z-offset value in the configuration.h at 0 and then change it on the LCD to -16.19?

TheMrCaveman, 2019-01-16:
Same here! Does not work after the G33 cleanup! @Ludy87 

This is quiete annoying. A lot has changed but the readme is not matching the current situation!

petieken, 2019-01-16:
I have this problem too since the latest bugfix I downloaded.

With the previous bugfix (a couple of weeks old) I had this problem didn't occur. The G33 function does not care about the Z-probe offset anymore. I tried setting it opposite (positive value instead of negative) and the result is the same.

I tried initialize EEPROM from the display, M502 & M500 in the console, and entering new offset with M851. None working.

KA8888, 2019-01-23:
This issue is solved, pls take a look at https://github.com/MarlinFirmware/Marlin/pull/12974.

closed 2019-01-23 by KA8888