MarlinFirmware/Marlin #12881: Can't figure out BLTouch

tommylee84, 2019-01-12:
### Description

I have an Creality CR-20 Pro and I've been trying to update the firmware to 1.1.9 from the 1.1.6 that came with it. I've followed all the youtube steps I could find and made sure that the pins and everything are set right. I can install TH3D just fine and the BLtouch works like a charm but every time I move back to the vanilla 1.1.9 the blue light never turns on and it doesn't respond via the LCD panel nor terminal commands. I've even added the TH3D files to compare. 
[Toms_TH3DUF_R2.zip](https://github.com/MarlinFirmware/Marlin/files/2751572/Toms_TH3DUF_R2.zip)

Board info: Creality 2.1 same as a CR-10s 
LCD: MiniPanel

[Tom_A_Marlin-1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/2751569/Tom_A_Marlin-1.1.x.zip)

**Expected behavior:** The BLTouch to work

**Actual behavior:** Nothing

RudyBenner, 2019-01-12:
Not really a firmware issue.

Better place to look is here.
https://www.facebook.com/groups/BLtouch/?ref=bookmarks
The first PINNED announcement message, right at the top, has everything you need.

rudy

mikeshub, 2019-01-12:
It took me a bit to figure out on MY CR-10s too. I had to completely removed my Z end stop sensor. Trying to use both was causing issues. Then you use the Z end stop sensor pin for the BLTouch.

The part that wasn't obvious to me was you want

`#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

and not

`#define Z_MIN_PROBE_ENDSTOP`

tommylee84, 2019-01-12:
So after a long night of stare and compare. I found this: 

#ifdef IS_RAMPS_13
  #define SERVO0_PIN        7   // RAMPS_13 // Will conflict with BTN_EN2 on LCD_I2C_VIKI
#else
  #define SERVO0_PIN       11

Since the CR-20Pro ID's as a RAMPS_13 it was setting 0 as pin 7 and no matter where I defined 0 in config.h it remained 7. So once I fixed that the BLtouch works but the blue light doesn't turn on like with TH3D but that's a problem for a different day.

shitcreek, 2019-01-23:
Glad that you've resolved this issue (at least partially).
Please close the issue. Thanks.

closed 2019-01-23 by thinkyhead