MarlinFirmware/Marlin #12888: HOTEND_OFFSET_ offsets lets carriage to drive past limits when not using T0

mikeshub, 2019-01-12:
### Description

Line 291 of motion.h position_is_reachable function needs to take into account which tool is active. The issue may take more than taking the nozzle offsets into account in that functions, but I'm in no rush to get on this bug. Maybe after my open PR gets looked at.

### Steps to Reproduce

* Define an X offset in Configuration.h

* Have T0 Active.

* Use G0 to command the carriage to go past the extents and observe it stop at the correct place

* Switch to T1 

* Use G0 to command the carriage to go past the extents and observe the machine drive attempt to drive the carriage past its physical limits along with all the angry noises.

**Expected behavior:** [What you expect to happen]

The defined min and max positions are to be respected regardless of the currently selected tool.

**Actual behavior:** [What actually happens]

The software disregards the nozzle offsets and drives the carriage past the machine's physical limits

#### Additional Information

The work around is to not use nozzle offsets in the firmware and instead put them into your slicer.

There is a similar issue when filament loading is done. If the bed is at its min the software lowers it regardless of whether a homing as taken place. This means that if the bed is already at the bottom it just keeps trying to drive it past the lowest extents of the bed.

mikeshub, 2019-01-14:
Fixed in PR 12855.

boelle, 2019-02-21:
@mikeshub problem solved?

mikeshub, 2019-02-21:
Not that I am aware of. My Creator Pro has been down due to smoking the mainboard. I hope to get it up and running again tonight now that my parts are finally in.

thinkyhead, 2019-02-23:
IF this isn't resolved, we'll get on this asap, as I know it has been a longstanding issue.

boelle, 2019-03-13:
@mikeshub had time to retest?

mikeshub, 2019-03-13:
Sorry no. When I destroyed the motherboard it took out the LCD. Replacement LCD finally arrived yesterday. I hope to test soon. Do you know if there was pull request made that fixes this issue?

boelle, 2019-03-13:
no i don't know

closed 2019-05-16 by thinkyhead