MarlinFirmware/Marlin #12898: MKS TFT28 communication error with MKS Gen1.4

rouhit110, 2019-01-13:
Hi friends
what parameter in marlin  should I use to use this LCD on this controller?
I recently enabled the filament runout sensor
When the filament finished, the extruder is transferred to the park's place, and some retract.. But nothing happens when I reordering filament
The LCD does not show any messages. The nozzle turns off and the temperature is not displayed on the lcd. pause andstop items not work.
generally, the lcd get interrupted with the controller

Now i uncommented #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

If anyone can help me, I will be very grateful

boelle, 2019-02-21:
@rouhit110 problem solved?

thinkyhead, 2019-03-03:
Please test with the `bugfix-2.0.x` branch so we're all on the same page here. That branch contains all the latest patches.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-05-16 by thinkyhead