MarlinFirmware/Marlin #12899: Problem with XXXX_PROBE_BED_POSITION on Anet A8

xgroelly, 2019-01-13:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2753123/Configuration.zip)

### Description

Hello everyone, 
I have problems with Marlin 1.1.9 (bulk and bugfix version). When setting the PROBE_BED_POSITION on an ANET A8 with BL-touch sensor and the Distant sensor mount from https://www.thingiverse.com/thing:2091529
It appears that coordinates of PROBE_BED_POSITION are coordinates of the NOZZLE and not the PROBE,  but on the other hand, the value validity asserting (during compilation) is done as if coordinates were related to the PROBE.
I attached the configuration.h file.

Have anyone an idea on the origin of the problem ?

FanDjango, 2019-01-13:
Had the same problem, long time ago, and so I decided to specify that explicitly:
```
  // Set the number of grid points per dimension.
  #define GRID_MAX_POINTS_X 7 //§§§§
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 15 //§§§§
  #define RIGHT_PROBE_BED_POSITION 205 //§§§§
  #define FRONT_PROBE_BED_POSITION 15 //§§§§
  #define BACK_PROBE_BED_POSITION 205 //§§§§
```
Didn't have time to analyze the logic behind the formulae in the original version - to find an error perhaps or just to understand why it is so quirky.

Of course, these positions must still be reachable by the probe and there is a sanity check on compile. I got the impression that the sanity check is doing the correct calculation.

Your probe offset might make (20,200)/(20,200) necessary.

Maybe someone needs to look at the formula.

boelle, 2019-02-21:
@xgroelly problem solved?

closed 2019-03-03 by thinkyhead

xgroelly, 2019-03-03:
The new bugfix firmware solved the issue