MarlinFirmware/Marlin #12901: request temp hotswap

jon012198, 2019-01-13:
can we get a EEPROM setting to change a temp sensor on the fly for testing so you dont have to keep re flashing firmware

GMagician, 2019-01-13:
Everything is possible but just to let you know how I think it works now:
temp table is compiled (it uses progam memory) so to swap to another temp sensor you need to keep both tables in memory (or worst all sensors tables in memory) and code should be changed to keep "last used sensor".
That means more program and sram memory just to an operation not done often...but maybe some other "more" skilled guy can give you a better response.

revilor, 2019-01-14:
It works like @GMagician said. I did something like that for my Delta printer which has interchangeable hotends. Each hotend module carries an NFC tag with information like nozzle diameter and the thermistor used. I modified Marlin to have up to 5 thermistor tables in progmem which is more than enough for my purpose (for all the 8 hotends I have 3 different thermistors in use at the moement). On startup or when the hotend is swapped the printer reads the NFC tag and activates the required thermistor table.

It's been some time since I implemented this and the whole solution contains NFC reader code and other stuff. But I can have a look if maybe I can factor out the thermistor part with not too much effort. But it's a very rare use case so I'm not sure if this would get merged into mainstream Marlin.

boelle, 2019-02-21:
@jon012198 problem solved?

closed 2019-02-21 by jon012198