MarlinFirmware/Marlin #12904: Problem Marlin 2.0 HEATER_0_PIN overlap

serdardogan, 2019-01-14:
### Description

Hi everyone,
I'm Teensy 3.6 user and I'm developing a shield right now. But in pins_TEENSY35_36.h, pin 30 is used as both E0_DIR_PIN and HEATER_0_PIN.

**Expected behavior:** I think HEATER_0_PIN pin 36 can be used

ajanky, 2019-01-17:
same problem here, just defining the Eagle file...
look at the comment graphics at the top of the file: E0 is moved to pins 17,18 and 19
However, I'm missing a Z Probe input - so if you have suggestions for added pins, lets share

serdardogan, 2019-01-18:
Pins in the upper part of the graph in the pin file and in the different definitions in the lower part. 
Do I have to accept the pins specified in the graph on the top or the pins on the bottom?

serdardogan, 2019-01-18:
thank you so much. I will follow the updates.

ajanky, 2019-01-19:
OK, you're pretty free to assign a new pin for the heater_0.
I have cleaned up the pin assignment for my purposes and started testing on a breadboard with scope. all outputs, endstops and Temp inputs work so far. Haven't tested display port yet. Tried to keep standard I2C and SPI port free. 
attached my current assignment. 
I will update after more tests.

[image]

serdardogan, 2019-01-21:
Thank you very much for helping. I can't wait to test it as soon as possible.

boelle, 2019-02-21:
@serdardogan problem solved?

serdardogan, 2019-02-27:
> 
> 
> @serdardogan problem solved?

I haven't checked in yet. I'il check it out as soon as possible. Sorry

serdardogan, 2019-03-09:
> 
> 
> @serdardogan had time to check?

Sorry.
I'm trying to install teensy now, but I've encountered a compilation error, I can't load it

serdardogan, 2019-03-09:
I am not getting the same error in my previous bugfix version.

boelle, 2019-03-13:
not tried with ide 1.9?

serdardogan, 2019-03-14:
I can't test arduino ide 1.9 because teensyduino is don't install for ide version 1.9.
I'm using and tested Arduino ide 1.8.8 and can't install to teenys 3.6

boelle, 2019-03-14:
and vscode+platformio does not support the teensy either?

serdardogan, 2019-03-14:
> 
> 
> and vscode+platformio does not support the teensy either?

I have no idea about this. I didn't use yet.

salfter, 2019-03-21:
> and vscode+platformio does not support the teensy either?

They do...haven't done much more yet than make the onboard LED blink faster, but the necessary compiler support is there.

serdardogan, 2019-03-21:
> 
> 
> > and vscode+platformio does not support the teensy either?
> 
> They do...haven't done much more yet than make the onboard LED blink faster, but the necessary compiler support is there.

I don't use teensy too much, I also bought it a short time ago. I'm sorry you acted like this.

closed 2019-05-16 by thinkyhead