MarlinFirmware/Marlin #12905: z-min endstop sense with piezo, after deactivated stepper G28, G29 errors.

njozsef, 2019-01-14:
Hi!

[Description]
I use Marlin 1.1.9 bugfix and ender3 with piezo (underbed) from precisionpiezo.uk.
if timed out steppers and disabled, G28 or G29 running error (printer halted) after z is when move down. Z-stepper big jerk first step, and big vibration after powered up.

I use 
`#define DELAY_BEFORE_PROBING 200   //on Configuration.h`

[reproducing]
1: power on
2: G28 or G29  (good, because unknown z, moving up z at 10mm to safe height )
3: move Z at +30mm (move highest position than safe height)
4: wait steppers timed out or M84
5: G28
6: activate x and y steppers and moving 0,0
7: moving bed center xy and  begins to observe z-min pin.
8: Big jerk az Z axis at first step down (powered up), piezo levelling fail-triggered z-min. Printer halted!!!

[suggestion]
Activate all motors at the beginning before running G28 or G29.

Thank you
Jozsef

njozsef, 2019-01-15:
piezo underbed version on thingiverse: https://www.thingiverse.com/thing:3348684
when it works G29 video: https://www.youtube.com/watch?v=CjpgHudlJSM

boelle, 2019-02-21:
@problem solved?

boelle, 2019-02-21:
@njozsef problem solved?

thinkyhead, 2019-03-03:
>  (printer halted) after z is when move down.

Lower your Z acceleration, maximum speed, and Z jerk values. And/or reduce the micro-steps on your Z stepper driver to 16x or 8x. This should prevent the Z axis getting torqued to the point of sticking.

njozsef, 2019-03-06:
Hi.
I understand, and accepted the lower acc and no jerk. But printer is working normally if motors not timed out.
Error only after motors  relased. After reconnect stepper H-bridge, high vibration generated. This is not related the acceleration and jerk.

Suggested solution is simple: **all motors powered on (include z) before G28.** 

currently firmware only powered on x and y motors before running g28. After xy moved the min point, powered on the z drive.

Thank you.
Jozsef

boelle, 2019-03-06:
maybe attach your 2 config files as a zip file, it could tell @thinkyhead what could be the cause

njozsef, 2019-03-11:
I solution (quick an dirty)
Setting (Marlin v2.0) one param. true to false
#define DISABLE_INACTIVE_Z false
Thank you. please close.

closed 2019-03-11 by njozsef