MarlinFirmware/Marlin #12906: EXTRA_FANSPEED - Possible improvement

MasterPIC, 2019-01-14:
I see that EXTRA_FANSPEED allows the user to set a secondary speed.
But during a print fan speed can change several times and, as currently implemented,  the secondary speed is not preserved. So I think that switching between fan_Speeds and old_fanSpeeds is not the best solution...

What about making
M106 T1 switching to the "usual" fan speed, the only one set by M106 Sxxx;
M106 T2 switching to the secondary fan speed;
M106 T3-255 setting the secondary fan speed.

M106 Sxxx should only change the primary speed, so that secondary speed won't be changed and actual fan speed won't change if the printer is in "secondary mode"...

Thanks
Armando

MasterPIC, 2019-01-14:
@thinkyhead I'm suggesting the following code:
```
void GcodeSuite::M106() {
  const uint8_t p = parser.byteval('P');
  const uint16_t s = parser.ushortval('S', 255);

  #if ENABLED(SINGLENOZZLE)
    if (p != active_extruder) {
      if (p < EXTRUDERS) singlenozzle_fan_speed[p] = MIN(s, 255U);
      return;
    }
  #endif

  if (p < FAN_COUNT) {
    #if ENABLED(EXTRA_FAN_SPEED)
      const int16_t t = parser.intval('T');
      if (t > 0) {
        switch (t) {
          case 1:
            secondary_fan_mode[p] = false;
            break;
          case 2:
            secondary_fan_mode[p] = true;
            break;
          default:
            secondary_fan_speed[p] = MIN(t, 255U);
            break;
        }
      }
      else
        primary_fan_speed[p] = MIN(s, 255U);
      fan_speed[p] = secondary_fan_mode[p] ? secondary_fan_speed[p] : primary_fan_speed[p];
    #else // EXTRA_FAN_SPEED
      fan_speed[p] = MIN(s, 255U);
  }
}
```
Because of the different solution I renamed **old_fan_speed** to **primary_fan_speed** and **new_fan_speed** to **secondary_fan_speed** to exclude confusion...
A boolean **secondary_fan_mode** array is required to store current fan modes.

boelle, 2019-02-21:
@MasterPIC submit a PR to bugfix 2.0

closed 2019-02-21 by MasterPIC