MarlinFirmware/Marlin #12907: Case Light on by default in menu but not actually turned on

stephenmott, 2019-01-14:
### Description

I've defined CASE_LIGHT_ENABLE and CASE_LIGHT_USE_NEOPIXEL and connected a short neopixel strip to pin 4 on an MKS GEN 1.3 board. Works fine, except initially when the board boots, it does it's NEOPIXEL_STARTUP_TEST, then goes off but the case light menu option says it's on. If I turn it off then on again in the menu, it goes on fine.

### Steps to Reproduce

1. Attach a neopixel strip to pin 4 on an MKS GEN 1.3 board
2. In configuration.h define:

```
// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN    4        // LED driving pin on motherboard 4 => D4 (EXP2-5 on Printrboard) / 30 => PC7 (EXP3-13 on Rumba)
  #define NEOPIXEL_PIXELS 5        // Number of LEDs in the strip
  //#define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 200  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup
#endif
```

3. In Configuration_adv.h define

```
#define CASE_LIGHT_ENABLE
#if ENABLED(CASE_LIGHT_ENABLE)
  #define CASE_LIGHT_PIN 4                  // Override the default pin if needed
  #define INVERT_CASE_LIGHT false             // Set true if Case Light is ON when pin is LOW
  #define CASE_LIGHT_DEFAULT_ON true          // Set default power-up state on
  #define CASE_LIGHT_DEFAULT_BRIGHTNESS 105   // Set default power-up brightness (0-255, requires PWM pin)
  #define MENU_ITEM_CASE_LIGHT              // Add a Case Light option to the LCD main menu
  #define CASE_LIGHT_USE_NEOPIXEL           // Use Neopixel LED as case light, requires NEOPIXEL_LED.
  #if ENABLED(CASE_LIGHT_USE_NEOPIXEL)
    #define CASE_LIGHT_NEOPIXEL_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
  #endif
#endif
```

**Expected behavior:**
It should obey the `CASE_LIGHT_DEFAULT_ON true` define

**Actual behavior:** 
It ignores the `CASE_LIGHT_DEFAULT_ON true` and is always off at startup

#### Additional Information

I've (hopefully) attached a zip of my configs.. This is with Marlin-bugfix-2.0.x downloaded on the 10th Jan, can't see an exact version, the boot screen just shows bugfix 2.0.x.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2755423/Config.zip)

GMagician, 2019-01-14:
Looking at code it seems to do that sure pin you use (4) is not defined for something else and on startup it is immediately reset?

GMagician, 2019-01-14:
Not sure but maybe `SET_OUTPUT(CASE_LIGHT_PIN);` is missing when `CASE_LIGHT_USE_NEOPIXEL` in function `update_case_light`

stephenmott, 2019-01-15:
Re pin 4, I've not set any custom pins on this printer, so I don't think so and it is just when it boots, works fine when manually turned off/on or during a print..

Re code, I tried making it:

```
  #if ENABLED(CASE_LIGHT_USE_NEOPIXEL)

    SET_OUTPUT(CASE_LIGHT_PIN);
    leds.set_color(
      MakeLEDColor(case_light_color.r, case_light_color.g, case_light_color.b, case_light_color.w, n10ct),
      false
    );

  #else // !CASE_LIGHT_USE_NEOPIXEL

```
but it's still off after boot..? I'm not exactly a C++ guy but could it be in `void LEDLights::setup()` this runs the `setup_neopixel` startup and then nothing, and the `setup_neopixel` turns them off if not `LED_USER_PRESET_STARTUP`, which I don't have defined, I don't even have `LED_CONTROL_MENU` defined..

stephenmott, 2019-01-15:
I think theres some confusion between LED defines and NEOPIXEL defines, if I add:

```
  #if ENABLED(CASE_LIGHT_ENABLE)
    if(CASE_LIGHT_DEFAULT_ON){
      set_neopixel_color(pixels.Color(255,255,255,255));    
    }
  #endif

```
to the end of `setup_neopixel` then it works as expected, well apart from the colours not using the defines..

GMagician, 2019-01-15:
I'll check. In setup there is a call to init case light, maybe such call is before neopixel is initialization

GMagician, 2019-01-15:
Please try my solution. If ok I'll post a PR here

stephenmott, 2019-01-15:
Thanks yep, that's working as expected now..

closed 2019-01-15 by stephenmott