MarlinFirmware/Marlin #12910: reprap smart lcd Marlin 2.0

davidcgu, 2019-01-14:
Good afternoon

I have an mks gen L main board with MKS TFT connected and one reprap LCD booth working at once.

Are running fine on 1.1.9 however I can not get running the LCD on 2.0 with same config than before.

I have un commented line:  #define REPRAP_DISCOUNT_SMART_CONTROLLER

Any advice?

kAdonis, 2019-01-14:
Are you sure you have a **RepRapDiscount Smart Controller**?
or do you have a **RepRapDiscount FULL GRAPHIC Smart Controller**?

davidcgu, 2019-01-14:
Is sure not a full grhaphic, is a 2004 with encoder & buzzer, I don't know how to attach here a picture.

Regards

marcio-ao, 2019-01-15:
> I don't know how to attach here a picture

Just drag-and-drop an image file onto the text-box.

boelle, 2019-02-21:
@davidcgu problem solved?

closed 2019-03-03 by thinkyhead