MarlinFirmware/Marlin #12917: Example configuration for Anet A8 has wrong X_MIN_POS and Y_MIN_POS values

black-zone-1000, 2019-01-15:
There is a very slight offset for X_MIN_POS (which is acceptable), but a huge offset for Y_MIN_POS (7mm offset).
The problem manifested when I added an inductive sensor in the right front corner of the extruder.
When executing G29 bed attempted to move to the maximum available position (position Y=220), but since the offset was wrong it crashed into the frame.

I measured carefully and those are the right values:
#define X_MIN_POS -34
#define Y_MIN_POS -3

P.S.: there is a slight deviation of less than one mm in both values, but when I tried float values the G29 command failed to execute, so I reverted back to integer values

RudyBenner, 2019-01-15:
Mine is

 #define X_MIN_POS -35
#define Y_MIN_POS -07

r.

FanDjango, 2019-01-15:
#define X_MIN_POS -14
#define Y_MIN_POS -3  
But I have bent my X-MIN-SWITCH outward somewhat to give the carriage a little more leeway on the left.
Originally my X_MIN_POS was 33.
There is some tolerance there for when you mount the switch with those little screws.
If you want the (110,110) position to be in exactly the bed middle, you need to correct these two values until you've hit the nail exactly.

FanDjango, 2019-01-15:
> When executing G29 bed attempted to move to the maximum available position (position Y=220), but since the offset was wrong it crashed into the frame.
And that would also be heavily influence by what offset you have specified for the sensor and what probe-points you have decided to use.
```
#define X_PROBE_OFFSET_FROM_EXTRUDER 28.0 // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -1.0 // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -2.6 // Z offset: -below +above  [the nozzle]
```
```
  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 15
  #define RIGHT_PROBE_BED_POSITION 205
  #define FRONT_PROBE_BED_POSITION 15
  #define BACK_PROBE_BED_POSITION 205
```
So on my machine, to probe front left, it would go to x = 205-28 + 14.  On the left side, x = 15-28 + 14, carriage very nearly at left boundary.

If you lie with the XXXX_PROBE_BED_POSITIONs, you will get a sanity_check fail on compile. But if the XX_PROBE_OFFSET_FROM_EXTRUDER values are wrong, a crash can happen, obviously.

black-zone-1000, 2019-01-16:
1. 
If the XX_PROBE_OFFSET_FROM_EXTRUDER are wrong, you should not see any crash, just the probing values will have offset resulting in a wrong bed leveling calibration.
2. 
I know you can always solve it using the left/right/front/back probe bed position, but... what will happen if you are printing something really big that reaches the y=219 coordinate of your heatbed... BOOM!
After applying the correct X_MIN_POS and Y_MIN_POS values, I could not crash the heatbed (or extruder) no matter what X and Y values I gave the G0 command due to firmware protection.

7eggert, 2019-01-17:
The example values are correct for my printer. (0;0) would be exactly on the edge if it wasn't rounded.

However, my Y axis crashed into the screws at the front - I shortened them. Fixed.

boelle, 2019-02-21:
@black-zone-1000 problem solved?

closed 2019-03-03 by thinkyhead

black-zone-1000, 2019-04-25:
> 
> 
> The example values are correct for my printer. (0;0) would be exactly on the edge if it wasn't rounded.
> 
> However, my Y axis crashed into the screws at the front - I shortened them. Fixed.

Well if you have to change the physical dimension of the printer then I would say this qualifies as the values are wrong...

7eggert, 2019-04-26:
The supplied screws are wrong - they need or face to the outside of the printer (which was later suggested to me) or to be cut (which I did). It's an error while assembling the printer due to having a bad manual.

TL;DR: Hardware issue, easily fixed.