MarlinFirmware/Marlin #12918: Using float values in MIN_X_POS and Y_MIN_POS fails the G29 Bed leveling command

black-zone-1000, 2019-01-15:
After accurate measurement I changed the configuration in Marlin with the following values:
define X_MIN_POS -32.5
#define Y_MIN_POS -2.8
The configuration successfully validated, and I have successfully updated the printers FW.

However, when I tried to perform a G29 instruction it failed with the following error (error shown in Octoprint terminal):
Recv: ? (L,R,F,B) out of bounds.

Returning the values to integer resolved the issue

RudyBenner, 2019-01-15:
#define X_MIN_POS -35
#define Y_MIN_POS -07

I think there will be variations between machines. The owner should expect to do a little work.

RudyBenner, 2019-01-15:
Yes, an integer is expected.

boelle, 2019-02-21:
@black-zone-1000 problem solved?

closed 2019-03-03 by thinkyhead

black-zone-1000, 2019-04-25:
@boelle  Is it possible to add a compile-time error instead of this very unsettling error when attempting to perform a G29?

boelle, 2019-04-25:
dont know, i'm not a coder

black-zone-1000, 2019-04-25:
So maybe change this to a feature request to add the validation?