MarlinFirmware/Marlin #12920: 1.1.9 Twi displays defined no error -- but

FlailAway, 2019-01-15:
I have an Anet A6 that the display has stopped working. It was getting dimmer and dimmer and now dark. I have a RepRap Smart display 20x 4 that I wanted to plug in.

I disable the #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and enable 
REPRAP_DISCOUNT_SMART_CONTROLLER and I get a compile error, 
"LCD_PINS_RS was not declared in this scope"

If I enable "FULL_GRAPHIC one as well as 20x4 it compiles fine, but the 20 x 4 does not work.

I found a thread in the 2.0 section but cannot figure out how to use that in 1.1.9.

Is it possible to use the 20 x 4?

Thanks

revilor, 2019-01-16:
#define REPRAP_DISCOUNT_SMART_CONTROLLER

seems to be the right choice. Problem is that there is no support in pins_ANET_10.h for this configuration which causes the compile error. To make it work you have to figure out which signal of the LCD goes to which pin on the board and define the LCD_PINS accordingly.

FlailAway, 2019-01-16:
Ahh, OK, thanks for the reply. I have found an Anet schematic so will see what I can do. Quite a ways out of my skill-zone, but would save me buying a new LCD setup.

Could I just copy the pin designation from a printer with the 20x4 LCD? The A8 has the 20x4 LCD is there some way I can tell Marlin to use the pins from that? I can't just define it as the A8 as the A6 has an encoder and the A8 uses 5-buttons. But, the main-boards in videos look to be the same. 

That's why I thought I could use the 20x4. I will check to see if there is an A8 sample config.

FlailAway, 2019-01-16:
OK, that did not work. Got a back-light but no characters.

boelle, 2019-02-21:
@FlailAway problem solved?

closed 2019-03-03 by thinkyhead