MarlinFirmware/Marlin #12922: [1.1.9] Retraction not possible with +600 steps/mm

bealparslan, 2019-01-15:
Hi, I use marlin **1.1.3 for my main printer over a year without any problem**. First of all, **_a sincere thank you to the development team.._** 

Last month I designed servo switch mechanism for bed leveling feature. Because, I’m tired of manual mesh leveling every day or week. And I downloaded, configured and **uploaded new Marlin 1.1.9**.
After some tuning, everything is perfect. **Now my prints are much better.**

And last week, I saw the Linear Advance thing. Sounds interesting and I tried. Because I have bowden system and sometimes corners little bad at high speeds (and this case, linear advance is my hero).

I follow official documentations on marlinfw.org. And I make script from this page: http://marlinfw.org/tools/lin_advance/k-factor.html
After starting the script, **extruder motor start makes horrible noise on retracts**.
I look at the script, everything is okay. Retract command is normal, speed normal, **everything as it's supposed to be in gcode**.

**In normal prints, same speed and acceleration, retraction works perfectly**. But alone G1 retract command, not working.
But, what is the reason? I don't know. Before opening the issue, let's make some research.
After 3 days of troubleshooting, I found the issue.
[image]
**(by the way, this problem starts after 1.1.8 upgrade)**

Problem is extruder instant accelleration. It does not accelerate instantly for my extruder.
Because I use **5:1 gear ratio extruder**. Generic marlin users are not affected because MK8 like extruder's steps per mm value around 95.
My **E-Steps per/mm value is 742.57** (1/16 microstepping) (by the way, **E3D Titan steps value is 800+**)
In this value, (based my research) maxium possible speed is ~21mm/s but I need 35mm/s! It's not possible, over the ~21mm/s, motor makes horrible sound.

Step values and max speeds based my research:
Used command: "G1 E3.9 Fspeed"
		

| Esteps  | Max possible speed |
| ------------- | ------------- |
| 742.57  | :no_entry:~**21mm/s** F1300 (I need 35mm/s!)(In 1.1.8, I up to 80mm/s!)  |
| 700  | :no_entry:~**23mm/s** F1400  |
| 600  | :no_entry:~**26mm/s** F1600  |
| 500  | :white_check_mark:~**78mm/s** F4700  |
| 400  | :white_check_mark:~**96mm/s** F5800  |

**Hardware information:**
COREXY, RAMPS_14_EFB, A4988

**Additional information:**
- I use A4988 driver. (I try **DRV8825** 1/32 double steps value: 1485.14 results proportional on A4988. **DRV8825 is little worse**.)
- I use default **3000mm/s extruder accelleration** everytime.
- I use 3.9mm retract with 35mm/s speed everytime. (G1 E-3.9 F2100)
- I try **releative** and **absolute**. Same results.
- I try both directions (1,-1). Same results.
- **I remove the motor from the extruder during all tests.**

FanDjango, 2019-01-15:
Why do you **need** 35mm/s, just a question?

I have limited my max E speed to 25mm/s on a 5:1 gear and measured the steps manually (normal step/mm = 100, 5:1 gear -> step/mm = 500, using 8825s, 4988s would be step/mm = 250), so:
```
#define DEFAULT_AXIS_STEPS_PER_UNIT   { xx, xx, xxxx, 500 }
```
You shouldn't be playing around with this value for your tests. It is a fixed value based on your manual measurement with a ruler, how many steps is needed for a millimeter of filament. Measure that once, put that into Marlin and leave it.

The things to play around with are FEEDRATE and ACCELERATION, maybe JERK.

I tested many speeds and above 30mm/s things got flaky, so:
```
#define DEFAULT_MAX_FEEDRATE          { xxx, xxx, x, 25 }
```

Thats about 25*500 = 15k steps/s, which I consider good for a small NEMA17 stepper. Someone correct me if theirs goes a lot faster with a gear and extruder as load. How many RPMs does that work out as, by the way?

**Anyway, your real riddle is, why does it work in everyday printing and in your manual test case, and why does it not work in the seemingly identical k-factor script?**

It shouldn't be working in any of these cases if your are exceeding your extruder/gears max capabilities with F2100, which is 2100 / 60 = 35mm/s.

So what is the difference between **Custom Test 1** and **Custom Test 2** ?

FanDjango, 2019-01-15:
Also, in your table:

If 742.57 steps are physically needed for a mm, then
```
742  F1300 works out to 21mm/s
700  F1400 works out to 21mm/s (because you are lying about the steps/mm)
600  F1600 works out to 21mm/s                  (ditto)
500  F4700 works out to 54mm/s                  (ditto)
400  F5800 works out to 52mm/s                  (ditto)
```
Why is there a "jump" in the table below 600 and those work at twice the extrusion rate and the ones above 500 are about half as fast?

bealparslan, 2019-01-15:
oh man!
First of all, 35mm/s is needed. 30mm/s **gives a lot of stringing**.
And, I know. Steps per/mm value is fixed for my extruder. **lol.**
Purpose in this table, just give perspective for developers; show these bad limits for different possible configurations.
Because version 1.1.8 not have this amount of speed drop on 500 to 600. This is a problem and reason why I opened this issue.

Test1 and Test2 difference; **1 is SPINING**, **2 is NOT SPINING**. 
Same acceleration, same jerk, same speed.
But Test1 have little bit pre movement. And pre movement solve this low speed problem. I don't know why. 
G1 E3 F4700 ; **It works on 1.1.8 but not 1.1.9** **why?**
For 1.1.9 we need 2 lines:
G1 E0.02 F100 ; simulates normal printing (pre movement)
G1 E3 F4700

> It shouldn't be working in any of these cases if your are exceeding your extruder/gears max capabilities with F2100, which is 2100 / 60 = 35mm/s.

[image]
Say hello to my 5:1 extuder, I used this over 1 years with 35mm/s retraction speed. Sometimes printing, sometimes custom gcodes, like filament load/unload at 35mm/s. I think it survives.

thinkyhead, 2019-01-16:
Please test the `bugfix-2.0.x` branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

bealparslan, 2019-01-16:
I upgrade firmware 1.1.9 to 2.0.x bugfix (last pull 12894) after some configuration, problem still available. Here is my configuration files:

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2762398/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2762400/Configuration_adv.zip)

FanDjango, 2019-01-16:
I have done some testing on my spare printer.

Marlin Bugfix-2.0.x from 2019.01.16

Szenario: Native serial terminal, no octoprint or any thing like that. Empty extruder/No filament. RESET, then M109 S190 to circumvent "cold extrusion", wait for heating... then:

Command used: G1 Fxxxx E100

First try:

DEFAULT_AXIS_STEPS_PER_UNIT 1000 (This is the correct value for my 5:1 geared extruder)
Highest possible was G1 F2300 E100. Any higher feedrate: Steps missing, buzzing motor
(F2300: that should be 38mm/s, E100 should extrude 100mm, so motor should run for about 3 seconds)

Second Try, which already corroberates what @bealparslan has discovered.

DEFAULT_AXIS_STEPS_PER_UNIT 500 (which is a lie, so E100 should extrude only 50mm, right?)
Highest possible was **again** G1 F2300 E100. Any higher feedrate: Steps missing, buzzing motor
(F2300: that should be 38mm/s, from the printers point of view it needs half as many steps in the same 3 seconds and then thinks it has extruded 100mm and actually just extruded 50mm). So the motor speed in that period should be half as fast, but if that were the case, you could request more than the F2300 without the motor failing. I would have expected F4600 to work. 

Third Try: DEFAULT_AXIS_STEPS_PER_UNIT 50 (Seems extreme, but what the heck...)
Any value, can't remember from what magnitude onwards, around 2000+, printer freeze, even RESET button on the contoller card doesn't work, need to power off and on.

Sadly, I forgot to use a stopwatch and longer extrudes (say 500mm) to get a better picture and some real filament in the extruder to check actual lengths. That's for tomorrow, maybe.

AnHardt, 2019-01-16:
Is `ADAPTIVE_STEP_SMOOTHING` active? If so - disable.

bealparslan, 2019-01-16:
> 
> 
> Is `ADAPTIVE_STEP_SMOOTHING` active? If so - disable.

It's default / disabled. [(I shared my configure files.](#issuecomment-454628408))
### In a nutshell

**After 1.1.8, marlin gives a lot of limitations on movements. It limits the printer's hardware capability. Especially on high eStep extruders. (+500 step; my extuder, e3d titan etc.)** 

@FanDjango Yeah you're right. DRV8825 is more worse than my results because I use A4988. Normally I use DRV but after this 1.1.9 update I feel something is wrong (motors feels like so weak/slow) and change it to A4988. A4988 is more better but still not enough. 

Also,
> Any value, can't remember from what magnitude onwards, around 2000+, printer freeze, even RESET button on the contoller card doesn't work, need to power off and on.

Yeah, **even with pre movement, printer does not move high speeds with long lengths (filament load/unload script), at result; it resets at the end of the movement.**

### Conclusion
_I'm waiting a fix for this problem. This motor limits are really annoying.
Hopefully, this problems fixed as soon as possible. I'm ready to share any kind of extra information of this issue, with developers._

gloomyandy, 2019-01-16:
Have you tried lowering the extruder acceleration settings? I would have expected that high steps/mm would require lower acceleration. I wonder if s curve acceleration would help (does it apply to the extruder?).

FanDjango, 2019-01-16:
I cannot reproduce this anymore. My 2nd test this morning must have had a firmware upload problem in front of the screen. I am sorry.

So I am out of this discussion, although it was a good opportunity to tweak my settings while testing this and here is what I did:

I was trying to get to the bottom of this.

Here's the original situation:
```
/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4[, E5]]]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT    { 200,  200,  800, 1000 } //§§§§
//                                                            // X,Y,Z,E0: DRV8825, 1/32 §§§§
//                                                            // E0: (1:5 Gear Extruder) §§§§

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4[, E5]]]]]
 */
#define DEFAULT_MAX_FEEDRATE          {  200,  200,   30,   36 } //§§§§
//                        (in mm/min) (12000 12000  1800  2160)    §§§§
//                                     =========================   §§§§
//                   ....(in steps/s)  40000 40000 24000 36000     §§§§
```
As you can see, the product **steps/mm * max_mm/s** is roughly 40000 for X and Y, 24000 for Z (plausible) and 36000 for E0. So far so good.

I tested the following with octoprint.

First, instead of playing with the extruder, I decided to do something that is visually easier to analyze:
I played with axis X, and lied that it had only 100 steps per mm instead of 200. I also allowed 400 max_mm per second. The product is still 40000.
To test this, I moved the carriage from X=1 to X=220 and back with F24000 (=400mm/s). No problem to use F24000 (twice my previous possible max), but the carriage only went to the middle of the bed, as expected. The speed of the move was the same movement speed as in the normal case, it was 200mm/s. But this proved that there is an inverse relationship between steps/mm and max_feedrate_possible.

So the X-Axis (and probably the Y-Axis and Z-Axis) are behaving as expected. Lowering the steps/mm allows a proportionally higher value for max_feedrate, and lowers the covered distance proportionally, the maximum achievable rotational speed of the stepper stays the same. 

Now I did the same for the extruder. I lowered the steps/mm from 1000 to 500 and increased the max_mm/s from 36 to 72. The behaviour was as to be expected: Half the extruded amount, same speed as before and now I was able to specify F4320 instead of F2160.

In none of these cases did I achieve a higher or lower motor speed.

Octoprint used G0 absolute for the X, Y and Z axis, and it used G91;G1 **relative**; G90 for the extruder moves.

So actually I cannot confirm the behaviour stated previously.

I also played with acceleration - if I lowered the accelaeration parameters of the extruder to a very low value, I could observe the extruder slowly ramping up its speed until it reached the maximum possible, after which the motor lost steps and made a loud sound.

The only thing left over from this, is the Marlin freeze when a very very low steps/mm and a very very high Fxxxxx ist used. I will test that tomorow.

gloomyandy, 2019-01-16:
"I also played with acceleration - if I lowered the accelaeration parameters of the extruder to a very low value, I could observe the extruder slowly ramping up its speed until it reached the maximum possible, after which the motor lost steps and made a loud sound."

Can you clarify that a little, are you saying that the speed increased but then while still accelerating it started to lose steps (so the target speed is too high), or are you saying that it accelerated up the target speed ok but then while trying to maintain that speed it lost steps? What was the target speed in these tests?

FanDjango, 2019-01-16:
@gloomyandy The speed increased slowly (as I had set acceleration to 50 only), and while accelerating it lost a few steps and immediately after that began to howl loudly, **before reaching the target speed**. I assume it reached the limits of it's capabilities - about 36-38 mm/s on 5:1 geared extruder (= target speed to high). Filament was in the bowden but bowden dangling in the air, so we had some load on the motor.

I also assume that without the 5:1 gear, we might achieve higher mm/s than 36. Because of the heavy load, however, we would need low acceleration and would have less than 5 times that speed. Cannot test this as no longer an ungeared extruder available.

FanDjango, 2019-01-16:
@bealparslan Sorry I didn't want to hijack your issue.
Back to the real question:
**So what is the difference between Custom Test 1 and Custom Test 2 ?**
Can you post your K-factor pattern GCODE script? And the values you entered in the form that generates it?

FanDjango, 2019-01-16:
> But Test1 have little bit pre movement. And pre movement solve this low speed problem.
Yeah, that would point to some kind of error in acceleration or jerk or whatever - but your pre-movement is so slow, funny that it would help. 
I looked at your config files and can see no real differences.

bealparslan, 2019-01-16:
### **I don't need acceleration for this. Because this is not fast a very fast movement actually. And, **again**, It works with older Marlin versions. Like 1.1.3, like 1.1.8.**
# Test your firmware!
Send these commands in order via _pronterface_ / _octopi terminal_ / _arduino serial_ / etc..
🔴🔴🔴 **Unload the filament from extruder before begin test**! 🔴🔴🔴

`M302 P1 ` 
`M92 E750`
`M201 E10000`
`M204 P3000 R3000 T3000`
`M83`
`G92 E0`
`G1 E5 F2100`
#### Do you hear horrible sound?
Configure and upload Marlin 1.1.8 or lower version and try again, problem disappears.

That's all.

bealparslan, 2019-01-16:
I found exactly similar issue [**like I mentioned on first post**](#issue-399520472), he is E3D Titan owner. But reasons are same, **With 500+ steps per mm value, your speeds limited so much (Titan estep value more than 800 actually, my extruder is around 750)**.. https://github.com/MarlinFirmware/Marlin/issues/11459

gloomyandy, 2019-01-17:
I think your test program may need a M203 E35 to raise the max feedrate for E (I think by default it is set to 25mm/s in 2.0.x). 

Not sure how useful my results will be but anyway...

I ran the test and it worked fine for me, no horrible noises. I added the M203 E35 and ran it again, the motor noise was now higher pitch (so the M203 was needed for my config), again ran fine. 

My system has a Titan Aero with a 0.9 degree stepper and I'm using 8825 drivers set to 32 microsteps so my extruder steps/mm is set to 1674, so to match your target speed I used M92 E1674. Ran the test again, even higher pitch motor noise, but again runs nicely, no nasty noise.

I'm running Marlin 2.0.x using an MKS Sbase board (so 32 bit). Maybe having the faster processor makes a difference?

FanDjango, 2019-01-17:
I have played around with the script. It extrudes. But I noticed a noise from the extruder **every second time I ran the script**. I've got a video with sound but the fan of the power supply is loud but one can hear it in the background. So I modified it to see/hear more (**please look at the comments in the gcode**):
```
M203 E50   ; be able to up to F3000 in the tests
M302 P1
M92 E1000
M201 E10000
M204 P3000 R3000 T3000
M83
G92 E0
G1 E15 F2100 ;this one extrudes very quietly
G1 E15 F2100 ;this one makes unpleasant noise but still extrudes ok
G1 E15 F2100 ;this one extrudes very quietly
G1 E15 F2100 ;noisy again
G1 E15 F2100 ;quietly
G1 E15 F2100 ;noisy again
G1 E15 F2100 ;quietly
G1 E15 F2100 ;and so on like meep meep meep meep while extruding
G1 E15 F2100
G1 E15 F2100
G1 E15 F2100
G1 E15 F2100
G1 E15 F2100
G1 E15 F2100
G1 E15 F2100
G1 E15 F2100
G1 E15 F2100
G1 E15 F2100
G1 E15 F2100
G1 E15 F2100
```
I managed to run this twice, after the second time of listening to the "meeep meeep meeep" while the extrusion was running, the controller is now frozen, the USB lamp is blinking rapidly, the reset button does not help, and I need to power off and power on the whole thing. Duh.

I also tried E5, but only two or three, this just makes the sequence faster: "mip - mip - mip".   

I also tried F2000. This produces "naat - naat - naat". 

**So every second extrude produces this extraneous rough sound, where you have the feeling it is not being driven properly. And if you are close to your maximum attainable speed, one extrude works, the next one blocks, the next one works again and so on.**

This happened for me at F2200 - One extrude will work, telling me that the motor CAN do it, the next one will block and make an awful sound and so on.
```
M203 E50
M302 P1
M92 E1000
M201 E10000
M204 P3000 R3000 T3000
M83
G92 E0
G1 E15 F2200 ;this one extrudes very quietly
G1 E15 F2200 ;this one makes unpleasant noise and DOES NOT extrude
G1 E15 F2200 ;this one extrudes very quietly
G1 E15 F2200 ;noisy again, no extrusion
G1 E15 F2200 ;quietly
G1 E15 F2200 ;noisy again, no extrusion
G1 E15 F2200 ;quietly
G1 E15 F2200 ;and so on
G1 E15 F2200
G1 E15 F2200
G1 E15 F2200
G1 E15 F2200
G1 E15 F2200
G1 E15 F2200
G1 E15 F2200
G1 E15 F2200
G1 E15 F2200
G1 E15 F2200
G1 E15 F2200
G1 E15 F2200
```

When I went up to F2400, continuous horrible sound, no extrusion.

**And every once in I while, total lock up of controller board, needs power off/on. Getting tired of that - it doesn't happen during normal operation.**

So @bealparslan: Back to the drawing board.

bealparslan, 2019-01-17:
> I think your test program may need a M203 E35 to raise the max feedrate for E (I think by default it is set to 25mm/s in 2.0.x).

Oh, thank you!
### **[Test script updated.](#issuecomment-454950047)**

> I ran the test and it worked fine for me, ..... I'm running Marlin 2.0.x using an MKS Sbase board (so 32 bit). Maybe having the faster processor **makes a difference?**...

**Yes it makes a difference**, try again with common 8-bit hardware like ramps+arduino mega if possible.

bealparslan, 2019-01-17:
Updated - Also M205 command added.

eightheads, 2019-02-07:
I'm running into the same issue currently with 1.1.9 running a LV8729 on a clone Bondtech BMG extruder. LV8729 is running at 1/32 and with the 3:1 ratio my esteps are currently ~825 esteps. Retracting at up to 45mm/s is fine, but anything faster than that gives a horrible sound and missed steps. Here's a video: https://youtu.be/QT0sjKY9QCA

AnHardt, 2019-02-07:
825 \* 45 = 37.125 KHz. That's about the expected limit for a 16 MHz 8 bit processor.

boelle, 2019-02-21:
@bealparslan problem solved?

eightheads, 2019-02-21:
> 825 * 45 = 37.125 KHz. That's about the expected limit for a 16 MHz 8 bit processor.

So this is more of a processor bottleneck rather than Marlin?

AnHardt, 2019-02-21:
Yes.
How about 1/16 microstepping? That would give you a god amount of headroom.

bogaskilop, 2019-02-23:
i observed that also on 1.1.9 bugfix  i was trying to change vrefs  but then realised  that vref are good   and everything  with retraction speed higher than 45 mm/s it makes a lot of noise and missed steps .  and this is from beginning in 1.1.9   I'm using  titan extruder  but what is interesting i got some 1.1.9 bugfix   marlin at work on one of machines and from few month back and on that machine i can use speed like 100 mm/s for retraction and all is good but when machine going to Z homing position (X297 Y300) tmc2130 on Y acting like   extruder in this issue .

i hope it is understandable

boelle, 2019-02-23:
not using 2.0?

bogaskilop, 2019-02-23:
nope can not compile it all the time getting some sanity check errors  lol

bogaskilop, 2019-02-24:
[thinkyhead](/thinkyhead)  can some one from developers  correct that issue with retraction speeds in 1.1.9 bugfix so at least last of 8-bit marlin version will work like it should. 
Everything what is above 1.1.6 is not working quite good :(  always something  wrong with things that was working good on older versions.

MARLIN 2.0 will not be ready to use for some time so it will be very good that last version of 1.1 will work properly with its inbuilt options.

boelle, 2019-02-24:
remember that bugfix 2.0 is both 32 bit and 8 bit

boelle, 2019-02-24:
@thinkyhead what he just said above ^^^

bogaskilop, 2019-02-24:
@boelle @thinkyhead   

So same setting  from working 1.1.9  moved to 2.0    and Z axis making a lot of noise   after that  movement  all motors making  super loud music .

So i can not test  that retraction speed on marlin 2.0 because of this :(

thinkyhead, 2019-03-03:
> and what is written here??

That's for the n00bs, not for you!

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead