MarlinFirmware/Marlin #12923: im haviung an issue with the graphics glitching out on my copy of marlin, more below

leytanelectronics2, 2019-01-15:
I have had my ramp's 1.4 printer kit for 2 days and finally got the thing working but the display is showing this weird distortion of the graphics seen in this video i posted on youtube ==> [**HERE**](https://www.youtube.com/watch?v=5jpvZ5Ux6KA&feature=youtu.be ) 

If anyone could tell me what is going on and how to fix this that would be really great because I am totally lost on this.

swilkens, 2019-01-16:
It would be helpfull if you posted your configuration files + let us know exactly which version of marlin you are using.

You might look through some of the various other issue reports that have similar LCD corruption issues. For example:

https://github.com/MarlinFirmware/Marlin/issues/9175

leytanelectronics2, 2019-01-17:
EDIT: i have solved the issue and went with a different version of marlin, using 1.1.9 and got a prebaked config file from github.

shitcreek, 2019-01-23:
Great. Please consider closing this issue. Thanks.

boelle, 2019-02-21:
@leytanelectronics2 problem solved?

closed 2019-03-03 by thinkyhead