MarlinFirmware/Marlin #12927: [Question]: Why does calling home_axis(Z_AXIS) in setup() cause axis to crash?

marcio-ao, 2019-01-16:
I would like to call `home_axis(Z_AXIS)` in setup() right before the servos in SWITCHING_NOZZLE are initialized.

The reason for this is that we've noticed that if the axis on our printer Z axis falls to the bottom during transport, it may push the nozzle up and if the servo is then initialized, it will then push against the bed, potentially burning out the servo.

I wanted to avoid this possibility by homing Z prior to energizing the servos:

```
void setup() {
  ...
  home_axis(Z_AXIS);
  #if ENABLED(SWITCHING_NOZZLE)
     move_nozzle_servo(0);     // Initialize nozzle servo
  #endif
  ...
}
``` 

However, the axis move all the way up but never stops, skipping the motor. This also happens if I put `home_axis(Z_AXIS);` at the very bottom of the `setup()` routine.

The following alternative works, but isn't as effective in protecting the servos as the servos are energized prior to the start of homing:

```
void setup() {
  ...
  enqueue_and_echo_commands_P("G28 Z");
  #if ENABLED(SWITCHING_NOZZLE)
     move_nozzle_servo(0);     // Initialize nozzle servo
  #endif
  ...
}
```

Any thoughts as to why my original code did not work?

marcio-ao, 2019-01-16:
**UPDATE:** The following seems to work and has the desired effect:

void setup() {
  ...
  GcodeSuite::process_subcommands_now_P("G28 Z");
  #if ENABLED(SWITCHING_NOZZLE)
     move_nozzle_servo(0);     // Initialize nozzle servo
  #endif
  ...
}

I guess it's still annoying that you can't execute `home_axis(Z_AXIS);` directly, but at least I have an alternative now that does the trick.

shitcreek, 2019-01-23:
Greate. Please consider closing this issue. Also if you are so inclined, maybe submit your solution to the project. Thanks

marcio-ao, 2019-01-23:
@shitcreek: I guess my question was never answered. It seems to me like "home_axis(Z_AXIS)" should behave like "G28 Z", but it does not. This appears to be a bug. That said, in so far as `process_subcommands_now_P` seems to be a viable alternative, and this issue affects only our printers, I will close this issue.

closed 2019-01-23 by marcio-ao

AnHardt, 2019-01-23:
`home_axis(Z_AXIS)` is just the actual homing of Z. `G28 Z` adds all the safety checks and preparations to make a safe `home_axis(Z_AXIS)` possible. For example it could be desasterous if you'd try to home the Z-tower on a delta without having all carriages near enough to the top, or the probe outside the bed, ... .

Rule:
Moving the machine before its state is known (homed) may crash.

Run your setup() after/during homing Z. You'll not be able to use the nozzles before homing anyway