MarlinFirmware/Marlin #12928: Z offset changing constantly with bltouch

Voteck, 2019-01-16:
We are running an Ender3 with a titan aero and a bltouch.
We correctly set the z offset for the nozzle and the prints work out great.
The problem is that the z offset value constantly changes. Today it was from -0.8 to 1.2, which is a lot.
We are making the adjustment on a hot nozzle so we take material expansion into consideration as well.
We always start a print with G29.

Why would the distance between the nozzle and bltouch change if they are on the same mount??

swilkens, 2019-01-16:
Is this a genuine BLTouch? Some of the knock-offs have been known to show variable measurements on repeat.

You can do a repeatabillity test on your probe with M48 to see how stable the output is?

http://marlinfw.org/docs/gcode/M048.html

Voteck, 2019-01-16:
https://www.amazon.co.uk/gp/product/B0748FJ48N/ref=ppx_yo_dt_b_asin_title_o01__o00_s00?ie=UTF8&psc=1

This is where we got it from. We will do the M48 when our print ends.

swilkens, 2019-01-16:
BIQU is an authorized re-seller (https://www.antclabs.com/store-1), that should be genuine.

This page inludes some configuration assistance if you haven't seen it already:
https://www.antclabs.com/bltouch

If the M48 gives decent results, please report back with your `Configuration.h` and `Configuration_adv.h` files. If possible, please also try to use the latest bugfix-2.0.x to verify if the issue still exists on the latest codebase.

Voteck, 2019-01-16:
But we do not have marlin 2.0. We do not have a mainboard that will use 32bit Marlin.
Should we install this?
1.1.x “bugfix” snapshot
Patches after 1.1.9

swilkens, 2019-01-16:
> But we do not have marlin 2.0. We do not have a mainboard that will use 32bit Marlin.
> Should we install this?
> 1.1.x “bugfix” snapshot
> Patches after 1.1.9

Marlin 2.0.x supports both 8 and 32 bit boards, it can still be used on your system and can still be compiled with the standard Arduino IDE 👍

Voteck, 2019-01-16:
Thank you for all of your help.
Do we just install the bugfix 2.0 file or do we need to update the Marlin first. Will this install on the Ender3 mainboard without any constraints?

swilkens, 2019-01-16:
Marlin 2.0.x-bugfix, which I linked above, can be installed in the same way as 1.1.9 or any of the other versions. The process is the same.

Do you have experience installing a new Marlin firmware from the base source on the Ender 3? If not, I would recommend this channel: https://www.youtube.com/watch?v=sUlqrSq6LeY

If you do not feel comfortable installing Marlin from source, you can also use the firmware that Creality provides:

https://www.creality3d.cn/download/firmware_c0001

But first try the test we discussed before.

Voteck, 2019-01-16:
Yes, we have experience in installing firmware. I was just asking because the name is misleading a bit. Like the file is not the Marlin 2.0 but a bugfix for it :)

We did run the M48 and those are the results:

Recv: Finished!
Recv: Mean: 0.001250 Min: -0.005 Max: 0.010 Range: 0.015
Recv: Standard Deviation: 0.003750
Recv:
Recv: X:97.00 Y:156.00 Z:11.10 E:0.00 Count X:7760 Y:12480 Z:4440
Recv: ok

So the diviation is fine. We will try the software update now.

Balzac40, 2019-01-17:
I have the same issue, with a **capacitive sensor**.
Two M48 runs 
20:08:14.392 : M48 Z-Probe Repeatability Test
20:09:30.774 : Finished!
20:09:30.775 : Mean: 0.030324 Min: -0.005 Max: 0.071 Range: 0.076
20:09:30.777 : Standard Deviation: 0.022813
20:09:30.781 : X:110.00 Y:170.00 Z:6.86 E:0.00 Count X:22000 Y:34000 Z:5490
20:10:25.127 : M48 Z-Probe Repeatability Test
20:11:42.278 : Finished!
20:11:42.279 : Mean: -0.085576 Min: -0.110 Max: -0.048 Range: 0.062
20:11:42.281 : Standard Deviation: 0.020713
20:11:42.285 : X:110.00 Y:170.00 Z:6.80 E:0.00 Count X:22000 Y:34000 Z:5443
I use 2.0bugfix and my Z-offset is varying between 1.85 and 2.00.
I often tweaks with M290.

Voteck, 2019-01-18:
> I have the same issue, with a **capacitive sensor**.
> Two M48 runs
> 20:08:14.392 : M48 Z-Probe Repeatability Test
> 20:09:30.774 : Finished!
> 20:09:30.775 : Mean: 0.030324 Min: -0.005 Max: 0.071 Range: [...]

Somebody said that this could be the whole mount for the extruder and motor that is loose and that it creates those kinds of divergences while slightly tilting due to vibrations of the printer. It could make sense because those are only .2 mm distances. I haven't got the chance to check this theory yet. I will let you know when I do.

boelle, 2019-02-21:
@Voteck problem solved?

Balzac40, 2019-02-22:
Mechanical issue for my printer. Seems to be solved

closed 2019-05-16 by thinkyhead

fer662, 2019-07-16:
If anyone runs into consistent z height increases on m48 like on one of the posts above, it might be lost steps or mechanic problems. If you want to rule out the probe being bad, re connect the z end stop you had before but keep the probe (i.e. just remove the 2 pin connector from the bltouch and connect the endstop). Do an m48 now and if you get inconsistent results as well, check for mechanical problems. I had a loose z rod. I didn't look loose at all but i was losing exactly 1 micron on each poll.

dburr, 2019-11-28:
@fer662 

> If anyone runs into consistent z height increases on m48 like on one of the posts above, it might be lost steps or mechanic problems. If you want to rule out the probe being bad, re connect the z [...]

OH. MY. GOD.

If only I had found this comment a week ago, it would have saved what little is left of my sanity (and my hair.)

I have been pulling my hair out trying to figure out why my probing was acting so wonky (consistent z height increases between probes.) After days and days of futile troubleshooting I came to the conclusion that my capacitive sensor was going flaky, so I ordered a BLtouch. That arrived this afternoon. Imagine my disappointment when, after installing the BLtouch, I saw the exact same behavior. Fortunately, this time, my Googling led me to this GitHub issue. Checked my Z rod and it didn't look or feel loose, but I reseated it anyway. And lo and behold, my probe works now!!!

THANK YOU for this comment. Hopefully anyone else having issues like this will happen upon this page.

fer662, 2019-11-28:
> @fer662
>
>
> OH. MY. GOD.
>
> If only I had found this comment a week ago, it would have saved what little is left of my sanity (and my hair.)
>
> I have been pulling my hair out trying to figure out why my [...]

Glad to be of help. It was surely tempting to blame the knockoff probe at first but in the end it's way more precise than it needs to be.