MarlinFirmware/Marlin #12930: MKS MINI12864 V2 contrast support

sartor, 2019-01-16:
### Description
Makerbase released new version of MINI12864 display (V2), based on UC1701
It have very low contrast by default. I've contact seller support and they gave me an instruction how to change contrast.

For some reason contrast setting is unavailable for MKS MINI12864 (variant of Minipanel) but board support software contrast regulation.

Please, add support to firmware for it.
Now contrast can be changed by editing u8g_dev_uc1701_mini12864.c fil in U8glib library on line 60:
`0x027,		/* contrast value */`
I've change it to 0x030 and now display is reading clearly

#### Additional Information
I think problem begins at line 351 of Conditionals_LCD.h:
```C
#if ENABLED(DOGLCD)

  #define HAS_LCD_CONTRAST ( \
      ENABLED(MAKRPANEL) \
   || ENABLED(CARTESIO_UI) \
   || ENABLED(VIKI2) \
   || ENABLED(miniVIKI) \
   || ENABLED(ELB_FULL_GRAPHIC_CONTROLLER) \
  )

  #if HAS_LCD_CONTRAST
    #ifndef LCD_CONTRAST_MIN
      #define LCD_CONTRAST_MIN 0
    #endif
    #ifndef LCD_CONTRAST_MAX
      #define LCD_CONTRAST_MAX 63
    #endif
    #ifndef DEFAULT_LCD_CONTRAST
      #define DEFAULT_LCD_CONTRAST 32
    #endif
  #endif
#endif
```

This setting of contrast on line 139 of Conditionals_LCD.h didn't change anything:
```c
#if ENABLED(MAKRPANEL) || ENABLED(MINIPANEL)
  #define DOGLCD
  #define ULTIPANEL
  #define DEFAULT_LCD_CONTRAST 17
#endif
```

arifbalik, 2019-01-24:
Hi,
Today I tried to fix the problem by adding following line in Conditionals_LCD.h
```c
|| ENABLED(MINIPANEL) \
```

Then I tried M250 command and it changed the contrast, but only made it completely white. I followed the code and it seemed normal. Then I changed DEFAULT_LCD_CONTRAST setting and made it default (0x027) but it also didn't work (completely white). 

So somehow u8g_dev_uc1701_mini12864_fn function in u8g_dev_uc1701_mini12864.c can't write registers properly, maybe parameters were wrong.

boelle, 2019-02-21:
@sartor sure this is a marlin problem?

sartor, 2019-02-22:
I'm not sure, but I've fixed it by modifying code of u8g library. I think we need to add contrast regulation to marlin in this way

boelle, 2019-02-22:
maybe @thinkyhead can say what is the best way... 

maybe attach the whole ug8 lib as a zip file so he can have a look when he gets time

RicardoEmePe, 2019-03-02:
Thanks, I'm very happy to resolve the problem wiht your help. I had a Tevo Flash with this issue. Now my Lcd is how it should have been from the beginning. Many thanks.

petieken, 2019-03-03:
I tried changing the contrast value in u8g_dev_uc1701_mini12864_HAL.cpp as the OP did but it didn't change anything for me unfortunately.

RicardoEmePe, 2019-03-03:
Sure,I'm not the most prepared person to answer you, but I'll test it. The correct file is "u8g_dev_uc1701_mini12864.c", and usually is inside yours documents subdirectoy, arduino, library. In my case  "C:\Users\Ricardo\Documents\Arduino\libraries\U8glib_Arduino-master\src\clib". Once you change the value to 0x030, save the file, compile Marlin and upload it to your printer.  It works if MSK MINI12864 has the ic 1701. Some times we have some libraries with similar name, be sure you change the value in the correct file and library. I hope I've helped

petieken, 2019-03-03:
Thanks a lot for the pointer! 
I'm using VScode and the file was in a different location. Under my documents there were only the files for the Arduino IDE, but I found the correct file under .piolibdeps. Changed the value to 0x030 and the contrast is fine now.

closed 2019-05-16 by thinkyhead