MarlinFirmware/Marlin #12932: [Bugfix 2.0] TMCStepper.h No such file or directory - Trying to upgrade from 1.1.9 to 2.0

Titi9237, 2019-01-16:
Hi,
### Description

I am trying to upgrade from 1.1.9 to Bugfix 2.0. I fill the parameters according to the 1.1.9 that was working well (see Configuration_1.1.x.zip). But when I try to compile the new Marlin.ino, I end up with a TMCStepper.h error.

I am using Arduino-beta 1.9 and the last Marlin 2.0 Bugfix
[Configuration_1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/2766008/Configuration_1.1.x.zip)
[Configuration_2.0.zip](https://github.com/MarlinFirmware/Marlin/files/2766009/Configuration_2.0.zip)

### Steps to Reproduce

1. Use the Configuration.h and Configuration_adv.h in attachments (the 2.0 version)
2. Compile

**Expected behavior:** Marlin should have been compiled 

**Actual behavior:** There is the following error :

> Arduino: 1.9.0-beta (Linux), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)" In file included from sketch/src/module/stepper.h:232:0,
                 from sketch/src/Marlin.cpp:36:
sketch/src/module/stepper_indirection.h:59:26: fatal error: TMCStepper.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

If you need any other info, feel free to ask.

Phr3d13, 2019-01-16:
Do you have the TMCStepper library installed?

Titi9237, 2019-01-17:
Yes

teemuatlut, 2019-01-17:
Note the difference between `TMC2130Stepper` and `TMCStepper` libraries.

closed 2019-01-17 by Titi9237