MarlinFirmware/Marlin #12936: the printer is slowing down and shaking/vibrating in some tight turns and small circles.- 32 bit

orydar, 2019-01-17:
I updated marlin_bugfix_2.0 two weeks ago.  I'm using it on Mks Sbase a cartesian printer and my slicing software is simlipfy3d.
most of the prints are going well.  in some curves, tight turns and small circles the print starts to vibrate and extremely slow down. this is the only situation I get step losing. not every curve is causing this problem.  in some situations, it can print a curve fast and smoothly in one layer and in the next layer, it can start vibrating for the same kind of curve.
I was trying everything that I can think of to fix this problem - acceleration, jerk, junction,  current, s curve,  buffers  Ark support and more. maybe I am miss understanding one or more of those features.

I saw the people talking about the STL resolution.  so I tried lowering the resolution and it didn't change this effect.
I created a vase that shows the problem.
I'm attaching a video, a G-Code, a bin file, an STL file and an image of the vase. in the image, you can see the texture that I get from this effect.
https://drive.google.com/drive/folders/1EaorFt-Q9Hb6DY8GabLonSED4J6GCr5Z?usp=sharing
I'm in the point that I use ridiculously low values of the jerk and acceleration.   
does someone have any idea what can cause this effect and how can I fix it?

I have another, smaller, problem referring to the EEPROM. in the new Marlin version, the EEPROM won't save to the SD card.  in the older version, I only needed to add a debug line.

swilkens, 2019-01-17:
Can you also share the `Configuration.h` and `Configuration_adv.h` files you used to compile this firmware?

Additionally, are you printing this directly from the SD card or are you using a serial interface (i.e. print over USB from S3D, ponterface, octoprint). If you are printing over serial, please try to print directly from the SD card.

AnHardt, 2019-01-17:
If using ADVANCE the constantly changing tool-width will confuse the algorithm. 
If printing over USB all these comments add unnecessary load.
When printing over USB and defining ADVANCED_OK you can observe if the planner buffer runs dry in the protocol. If so reduce the STL resolution even more.

gloomyandy, 2019-01-17:
With respect to the EEPROM question, what problem are you actually having? Are you saying that EEPROM data is not being saved at all? Recent versions of Marlin for the SBase no longer use a file on the SD card be default, they emulate EEPROM in flash memory. But the data should still be being saved/restored. There was a bug with this a month or so ago, so you may need to update if you are not using a very recent version of 2.0.x.

orydar, 2019-01-17:
thank you for your quick reply.
this is nuts and USD issue.  I am working only with an SD card.
the G-Code is not closed as a vase mode.
the links below are for the configuration files:
https://drive.google.com/file/d/1F8mn-TJNmyi6h_bV5vy2tfbAGU-4FmSF/view?usp=sharing

https://drive.google.com/file/d/1XGFBkIuZyRecInJilScQyWJm0oIizl7s/view?usp=sharing

 about the tool change width

 ,  I'm checking it out. did you refer to the perimeters that are in the picture?
[image]

orydar, 2019-01-17:
the test of changing  the Extrusion width to be always 100%  didn't solve the problem.

orydar, 2019-01-17:
referring to the EEPROM.  the problem is it's not saving at all.  I am using M290 Z0.05 and M290 Z-0.05  to do a live calibration.  in the past, it saved it in the EEPROM.  today I need to do a live calibration every print and it does not save the EEPROM file to the SD.
I heard that there was a bug in the EEPROM  until a month ago.  my firmware is from two weeks ago.  if there is a reason I will update it again.

gloomyandy, 2019-01-17:
As I said above the EEPROM is not saved to SD any more it uses flash memory. So there is no point in looking at the SD. What are you doing to save your settings and how are you checking to see if they are saved? You should try it again with a current version of Marlin. You may also want to try changing other values in the eeprom to see if those values are stored and retrieved correctly. 

There may have been changes in the way that M290 works (I don't use it so can't say), so you may want to check that (in particular it looks like you need to use m500 to save the values). If you still think there is a problem you may want to open a separate issue so that more people see it.

BrienAllison, 2019-01-18:
I was having the exact same issue although to a larger degree than you. At small circular corners it would slow down massively, sometimes pausing momentarily and the print quality was really bad, rough due to the slow jerky motion. I had no idea what was causing the problem until @swilkens mentioned the command buffer. I am printing over USB but I had been doing it for a while without any problems, then I started having getting them about a month ago.

It turns out that my newly installed webcam was slowing down my raspberry pi a lot, It may be the type of camera which doesn't process the camera images or whatever (its an old playstation move camera I had laying around).      

Also I think my raspberry pi setup probably compounded the issue. I run my prints off a Raspberry Pi zero which I connect to the printer through a USB hub connected to the micro USB of the Pi. The hub also has an ethernet connection which I use to connect to the Pi. The webcam can't run off this first hub however because the Pi can't deliver enough power for it so it's connected via a secondary powered USB hub plugged into the first. It's a real mess and I'm guessing all those signals going through that one micro USB port will bottleneck everthing plus the Pi needing to process the webcam image caused it to not keep up with the printer at these high detail areas.

The picture shows the difference in print quality. They are the exact same Gcode with the webcam turned on for the part on the right and off for the part on the left, massive improvement!  
[image]
Slowing the outer shell speed all the way down to 10mm/s made it work with the webcam, but is not a great solution. Looks like I'll have to check up on my prints manually for the time being.

I'm wanting to get a normal Pi so I can get rid of all my hubs and make it much neater and then hopefully it will also work properly with the webcam. I could maybe install an SD reader to m printer although I have TMC 2130s currently using SPI so I don't know if that may also cause a bottleneck.

boelle, 2019-02-21:
@orydar problem solved?

closed 2019-03-03 by thinkyhead