MarlinFirmware/Marlin #12937: [bugfix-2.0.x] ULTRA_LCD + REPRAP_DISCOUNT_SMART_CONTROLLER is buggy

mordaha, 2019-01-17:
### Description

I connected standard  RepRap Smart LCD Controller to my DUE_USB+RURAMPS4D_13 (Marlin bugfix-2.0.x 32 bit) with this config:

```
#define REPRAP_DISCOUNT_SMART_CONTROLLER

#define ULTRA_LCD
#define NEWPANEL
#define LCD_WIDTH 20
#define LCD_HEIGHT 4
```

And it didn't work, it shows 2 strings of trash, something like this: https://i.imgur.com/RtGtHNw.png 

I debugged a little through the code and found that `LiquidCrystal lcd()` somehow initialized wrong or not initialized at all.

So, here, at L900 (https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/HD44780/lcdprint_hd44780.cpp#L900) in `lcd_put_wchar_max()`, `lcd.` is useless (pointing to a wrong device, not my?).

I added initialisation (hack) just before that `if`:
```  
  // initialize lcd()
  LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); 

  // TODO: fix the '\\' that doesnt exist in the HD44870
  if (c < 128) {
    lcd.write((uint8_t)c);
    return 1;
  }
```
And it started to work fine, as expected: https://i.imgur.com/incMr9w.png

I didn't debug through that lot of `#if ENABLED() -> #define` symbols, so i don't know where exactly the buggy code is, but i think it somewhere in defines/configs. May be lcd_i2c config somehow fired.

My full Configuration.h: https://pastebin.com/Q6J85X2k

How else can i help?

gloomyandy, 2019-01-17:
I'm not sure if this is causing any of your problems but you should not have to define:
```C
#define ULTRA_LCD
#define NEWPANEL
#define LCD_WIDTH 20
#define LCD_HEIGHT 4
```

Normally just defining REPRAP_DISCOUNT_SMART_CONTROLLER should be enough, all of the other defines should get set automatically if they are required. Was there a reason for adding them explicitly?

mordaha, 2019-01-17:
I googled it before install but i didn't find in code where REPRAP_DISCOUNT_SMART_CONTROLLER defines ULTRA_LCD, without UTRA_LCD it just not defines any pins (https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_RURAMPS4D_13.h#L189)

my bet - the devil in defines, i am sure)

mordaha, 2019-01-17:
(translating): with only #define REPRAP_DISCOUNT_SMART_CONTROLLER  it do nothing

gloomyandy, 2019-01-17:
You should look at Conditionals_Lcd.h
If you look through that you should be able to see what gets defined. You really should not have to define the individual components.

mordaha, 2019-01-18:
Again, i started from that point.
At first I defined only REPRAP_DISCOUNT_SMART_CONTROLLER and it didn't work. Blank screen.

gloomyandy, 2019-01-18:
Yes I understand that, but that does not mean you need to define those settings to get things working. It is more likely that there is a problem with the pins file for your board and that the pins are not set correctly for the display you are trying to use.

mordaha, 2019-01-18:
There is definitely bug in configs so i opened this issue!

gloomyandy, 2019-01-18:
The thing you need to remember is that you are trying to use a new version of Marlin with new 32bit hardware. This new version does not always support all of the same displays etc. that the older 8 bit version did on all of the different control boards. In this case I suspect that no one has actually implemented support for the display you are trying to use on the control board you have.

In some cases (like the smoothieboard based boards (like MKS Sbase), it may be that it is simply not possible to support some of the displays. I'm not sure if that is the case with your hardware, but in the case of the MKS SBase board (which I'm more familiar with), the control board simply does not have connections from the cpu to the connectors used for the display for some of the signals required by a display like the one you are trying to use. But other displays (like the RepRapDiscount Full Graphic Smart Controller), which require different signals work fine. It may be that you need to use a different type of display with the control board you have chosen.

mordaha, 2019-01-18:
I dont wanna somewhat hate you pal! ) i did it work and think that my experience will help you! nothing more) 
it is works for me so i'd like to find the core of a problem with you to help all others users, plz don't hate me too!
I can do any experiments for you service!

gloomyandy, 2019-01-18:
As I said you really should not have to set those defines yourself. How old is the 2.0.x you are using? Looking at the current code as far as I can see if you define REPRAP_DISCOUNT_SMART_CONTROLLER then the code in Conditionals_LCD.h should be defining ULTRA_LCD and NEWPANEL for you. In this file we have the following lines:
```C
#if ENABLED(ULTIMAKERCONTROLLER)              \
 || ENABLED(REPRAP_DISCOUNT_SMART_CONTROLLER) \
 || ENABLED(G3D_PANEL)                        \
 || ENABLED(RIGIDBOT_PANEL)                   \
 || ENABLED(ULTI_CONTROLLER)
  #define ULTIPANEL
#endif
```
then later in the same file there is
```C
#if ENABLED(ULTIPANEL)
  #define NEWPANEL  // Disable this if you actually have no click-encoder panel
  #define ULTRA_LCD
#endif
```
So as far as I can see ULTRA_LCD and NEWPANEL should be defined. So if you have the time the thing to understand is why they are not defined for you.

gloomyandy, 2019-01-19:
It may be worth getting a more recent version of 2.0.x. You never know there may have been a bug introduced and fixed.

claymore666, 2019-01-28:
Forgive me to hook into that conversation, but I think I have stumbled across a related issue and I too am thinking there is something wrong in the config.
I am trying to get an MKS GEN L running with Marlin 1.1.9 and the Ender 3 (ie CR10_STOCKDISPLAY to my knowledge) up and running - with one ribbon cable.

When reviewing the pinout I discovered that 
_#define CR10_STOCKDISPLAY_ will never set any pins for me as the _pin_ramps.h_ states:

`
#if ENABLED(ULTRA_LCD)

  //
  // LCD Display output pins
  //
  #if ENABLED(REPRAPWORLD_GRAPHICAL_LCD)

	...
  #elif ENABLED(NEWPANEL) && ENABLED(PANEL_ONE)

    ...
  #else

    #if ENABLED(CR10_STOCKDISPLAY)

      #define LCD_PINS_RS       27
      #define LCD_PINS_ENABLE   29
      #define LCD_PINS_D4       25

		...
    #endif

  
  #endif
`
Is it intentional to have set ULTRA_LCD in order to use CD10_STOCKDISPLAY? I guess thats also where @mordaha  is comming from...

boelle, 2019-02-21:
@mordaha problem solved?

mordaha, 2019-02-22:
"And it started to work fine, as expected: https://i.imgur.com/incMr9w.png"

for me - yes, not for marlin sources

Miran81, 2019-03-06:
@mordaha  how it works ? photo links broken. I'm using due+ramps-fd and 2004 lcd. Lcd is not working. Just squares on screen.

boelle, 2019-03-06:
> "And it started to work fine, as expected: https://i.imgur.com/incMr9w.png"
> 
> for me - yes, not for marlin sources

are you sure the problem is still here with todays copy of bugfix 2.0? have you tested?

Miran81, 2019-03-07:
@boelle  ı think its about ULTRA_LCD but ı don't know

mordaha, 2019-03-09:
>are you sure the problem is still here with todays copy of bugfix 2.0? have you tested?

i can in git ))

[image]

mordaha, 2019-03-09:
>@boelle ı think its about ULTRA_LCD but ı don't know

i think the problem is in that lot of defines, somewhere they are failed, but i don't know how to debug it :(

Miran81, 2019-03-09:
@boelle you sad your lcd is work fine. How ?

boelle, 2019-03-09:
my full graph display on re-arm+ramps ??

yes works fine

Miran81, 2019-03-09:
@mordaha you sad work fine. How ? I have ramps-fd+due 2004 lcd

mordaha, 2019-03-09:
@Miran81 i had explained it in the first message)

just added an initialization of lcd in the print function

Miran81, 2019-03-09:
@mordaha can you share configuration.h please :)

Miran81, 2019-03-09:
@mordaha i did it   // initialize lcd()
  LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7);  and look at picture
[image]

Miran81, 2019-03-09:
problem solved thank u all  💯  👍 👍 👍

closed 2019-05-16 by thinkyhead