MarlinFirmware/Marlin #12942: Unknown command: "M122"

sponge14, 2019-01-18:
I have set my XYZ and E0 driver types to TMC2208 in configuration.h  I have read where I am supposed to uncomment a line that says HAS_TRINAMIC, or something like that, but I cannot find that line in the latest 2.0 code.  If found HAS_TRINAMIC already defined in the Drivers.h file.

I am using a RE-Arm board with a RAMPS 1.6 board

When I try to send an M122 in Octorprint, I get the error Unknown command: "M122"

It seems the code is different now than has worked in the past?  Or am I just missing something here?

[Configuration - Copy.txt](https://github.com/MarlinFirmware/Marlin/files/2771589/Configuration.-.Copy.txt)
[Configuration_adv - Copy.txt](https://github.com/MarlinFirmware/Marlin/files/2771590/Configuration_adv.-.Copy.txt)

Msq001, 2019-01-18:
uncomment #define TMC_DEBUG in Configuration_adv.h

sponge14, 2019-01-18:
TMC_DEBUG was already disabled, sorry I missed saying that earlier.

teemuatlut, 2019-01-18:
`TMC2208_STANDALONE` tells Marlin that you do **not** have the communication lines wired to the driver and wish to run the driver in standalone mode. This only tunes the pulse timings to be suitable with the driver but does not enable any of the commands or features supported by the firmware, such as `M122`.

If you have wired the communication lines to the driver(s), use the correct option `TMC2208`.

> I have read where I am supposed to uncomment a line that says HAS_TRINAMIC

Don't do that.

> uncomment #define TMC_DEBUG in Configuration_adv.h

If running latest bugfix-2.0.x this is no longer necessary for `M122`.

sponge14, 2019-01-28:
No, I haven't yet been able to resolve the issue.

boelle, 2019-02-21:
@sponge14 problem solved in bugfix 2.0?

closed 2019-03-03 by thinkyhead