MarlinFirmware/Marlin #12945: [bugfix-2.0.x] Power loss recovery with PSU control

Titi9237, 2019-01-18:
Hi,

### Description
I just tried the Power loss recovery option using a MKS220v DET. I also have the PSU control option. 
It is detecting a powerloss as expected but when I put the power back on, I can choose to restart the print but it does not restart the PSU.

Titi9237, 2019-01-21:
If is a Corsair 550W power supply with the green wire connected to the PS_ON pin.
I enabled the PSU_Default_Off option.

boelle, 2019-02-21:
@Titi9237 problem solved?

Titi9237, 2019-02-22:
Nope, but I have not tried the function since then.

boelle, 2019-02-22:
maybe give it a go with latest copy of 2.0?

AnHardt, 2019-02-22:
Maybe fixed by https://github.com/MarlinFirmware/Marlin/commit/89bbc30a09b854c1fa20f63113a92c33df65e19a

Titi9237, 2019-02-22:
In theory, it seems to be the solution. Thanks !
But I will not be able to test it for the moment.
You can close the issue if necessary.

boelle, 2019-02-22:
we cant close, only you and @thinkyhead can do so

closed 2019-02-22 by Titi9237