MarlinFirmware/Marlin #12946: Print Counter: total time printing not updating (1.1.8 and bugfix2.0.x)

Luis-Lange, 2019-01-18:
### Description

Total time printed is not been updated when it is printing, only when is warming up.

### Steps to Reproduce

Even on LCD or M78 values are equal. If I start manually via M75, timer is increased.

**Expected behavior:** Increasing timer when printer is printing

**Actual behavior:** only increase timer when printer is warming up

#### Additional Information

I have same behaviour with 1.1.8 and bugfix2.0. Attached is configuration files for both versions.
[Configuration_1.1.8.zip](https://github.com/MarlinFirmware/Marlin/files/2773955/Configuration_1.1.8.zip)
[Configuration_bugfix2.0.zip](https://github.com/MarlinFirmware/Marlin/files/2773956/Configuration_bugfix2.0.zip)

Luis-Lange, 2019-01-25:
No tips?

InsanityAutomation, 2019-01-25:
Do you have autostart print timer enabled in configuration.h?

boelle, 2019-02-21:
@Luis-Lange problem solved?

Luis-Lange, 2019-02-22:
Hi Bo ! Thanks for asking. I was out of business due to some broken parts in my printer and today I could test again. I re-flash bugfix 2.0 (same I have in my computer, no new version) and additionally clear all the EEPROM contents (via a loop in Arduino IDE) and now it is working, both via SD card and OctoPrint.
I can't tell what solved, but for now it is working.

boelle, 2019-02-22:
then maybe click close?

Luis-Lange, 2019-02-22:
Sure ! Thanks !

closed 2019-02-22 by Luis-Lange