MarlinFirmware/Marlin #12947: [1.1.9] BLTouch suddenly stops working

Joshude, 2019-01-18:
Hello! I'm new to this whole github thing but I'm not new to the internet, so i hope i'll do anything to make my description as good as possible

### Description

first of all, here are my configs: 
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2774165/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2774166/Configuration.txt)

- I'm running Marlin 1.1.9 with BLTouch installed (to pin 27, disabled the buzzer, cut wires)
- BLTouch worked just fine with no problems for few weeks (while i'm new to 3D-printing and my Ender 3 is frustrating me so i'm hoping to get this issue fixed because if not i might throw that thing out of the window :p )
- If i home all axis X and Y just work as expected. If it comes to Z in like 90% the Z axis lifts a bit, deploys the BLTouch, then stores it, the display of my Ender 3 then just says "STOPPED", Octoprint says "STOP called because of BLTouch error - restart with M999" (yes, i found dozens of issues to this but nothing worked for me or i didn't understood the suggested solutions) and that was it. 
- BLTouch is geniue (i think, bought as such), so no clone
- Self-test just works fine (initial deploys and stows 2 times, the manual test just keeps "clicking" (as expected)
- M119 always returns triggered for Z-axis, but if i got it right it is "as designed"

I don't really what you need to know too but don't hesitate to ask.

FanDjango, 2019-01-18:
> M119 always returns triggered for Z-axis, but if i got it right it is "as designed"

Hmmm. Are you sure that should be like that? Where did you read that/get that from?

Your #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN says it should use the z min input and that means it certainly should not always be triggered.

Is the three-wire trio still connected well to the z-min port on the controller board? 

That you can deploy and self test confirms your servo pin is correctly wired up and consistant with the config.

So what changed from before when it worked?

Oh, raise the Z axis by manually screwing it up to about 100cm. Then deploy the probe, you can user the marlin display or octoprint bltouch plugin. Deploy, check M119, then push the litte plug up with your fingers and check M119 again. If it now goes the opposite, you need to changed the polarity in the config and ask yourself how it got set wrong some time ago from when it still worked.

Joshude, 2019-01-18:
> > M119 always returns triggered for Z-axis, but if i got it right it is "as designed"
> 
> Hmmm. Are you sure that should be like that? Where did you read that/get that from?
> 
Not sure right now, since i can't find where i read this. Was something like the value is pulsing and it's more or less coincidence if you see z_min triggered or open. But mabye i misundersood something.

> Is the three-wire trio still connected well to the z-min port on the controller board?
> 
I've checked them before posting. The connector seemed to be a bot off the socket but even after fixing that the issue is not gone

> That you can deploy and self test confirms your servo pin is correctly wired up and consistant with the config.

Thats what i've thought, too

> 
> So what changed from before when it worked?

Nothing, really. Thats why i'm confused. I mean, maybe my BLTouch just went dead, idk. 
> Oh, raise the Z axis by manually screwing it up to about 100cm. Then deploy the probe, you can user the marlin display or octoprint bltouch plugin. Deploy, check M119, then push the litte plug up [...]

Thanks, will test that.

Also, just updated to the [Bugfix]1.1.x, set the configs as above, flashed. First it looked good, but on the 3rd or 4th point of measuring (5x5 grid) the "pin" went up, down and up again, stayed there and octoprint stated "Probing failed"

FanDjango, 2019-01-18:
So homing Z worked. Observe the Z homing. pin goes down, nozzle moves down until pin touches bed, pin gets pulled up and nozzle goes up again a little. Then a slow one downwards with pin extended, touches bed, pin goes up and nozzle goes up again. Same thing should be happening in sequence on G29 levelling.

Joshude, 2019-01-18:
Tested the BLTouch as you said, @FanDjango 
Results: 
- Initially after rebooting the Ender 3 (and the selftest "clicks" two times: M119: z_min: open
- "probe down": M119: z_min: open
- triggered by finger: probe "clicks" up, light is red, probe "clicks" down, z_min: open
- "probe up": z_min: open

so... now my state is always "open", nevertheless what i do. 

> So homing Z worked. Observe the Z homing. pin goes down, nozzle moves down until pin touches bed, pin gets pulled up and nozzle goes up again a little. Then a slow one downwards with pin extended, [...]

this is exactly what's happening until (at this time) the 3rd or 4th probe where it just stopps.

Joshude, 2019-01-18:
Did the G48. Worked okay'ish. I had a hard time with "G28 Z", the probe was just going down and up again. Just after a "full" G28 Z-homing worked. When sending G48 the first probe was okay, the second "failed" (probed, went up, 2-3 sec pause and probed again). The others were okay. Results: 

```
Recv: Finished!
Recv: Mean: 0.039000 Min: 0.025 Max: 0.055 Range: 0.030
Recv: Standard Deviation: 0.010320
```
Will try your other tweaks asap.

FanDjango, 2019-01-18:
Sounds really flaky. The other "tweaks" won't help you. Wouldn't it be nice to have a second BLTouch to cross-check? Works / Works not kind of makes you hope its not broken. Totally check the cabling - I've had one case (it was the bed thermistor) that had intermittent contact depending on the movement of the bed.
In this case, it would be the movement of the nozzle carriage.

Joshude, 2019-01-18:
Yep, was going to say that i'll redo the cabling tomorrow. This is just too random to find another cause for that issue. 
Will report back with results. A huge thanks so far, @FanDjango !

FanDjango, 2019-01-18:
Oh, that gives you the chance to remove the probe from the carriage and to do the G48, but using your finger to "simulate" the touches. Thereby the cable is far awy from the other moving stuff and interference. Make sure to put a bar of soap or a marshmallow under the nozzle in case you fall asleep doing th simulated touches with your finger or drop the probe because that cat came in

Joshude, 2019-01-18:
*sigh*
one last post for today: 
Did a G48, worked. Homed several times, worked. Homed just G28Z, worked. Started a print, bed and nozzle were heated, the G29 started... and immediately failed. 
This is just stupid and random. 

But, enough for the day, i'll be back after recabling!

reloxx13, 2019-01-18:
u dont need to g29 every print.
use m420 s1 in ur startcode after g28 to enable the mesh or enable RESTORE_MESH_AFTER_G28 in ur fw config.

to level once
g29 ;abl
m500 ;save
m501 ;load

Joshude, 2019-01-21:
Rewired everything and now the BLTouch is working properly. Now i have other issues, but this seems to be resolved!

Thanks, @FanDjango 
Also thanks @reloxx13, implemented M420 S1 into my startcode!

boelle, 2019-02-21:
@Joshude problem solved?

closed 2019-03-03 by thinkyhead

100ccmREFO, 2019-04-01:
Hi @Joshude 
I have the exact same problem as you descripted it here.. Why did the rewiring help? Did you find some broken wires?

Thanks for your reply because i really tried everything and don't know what to do anymore..

dixi83, 2019-11-14:
For other people who are reading this, this problem is (almost certainly) probably caused by [interference](https://en.wikipedia.org/wiki/Electromagnetic_compatibility) of the z stepper . If you check the wiring docs of BLtouch there is a note that says:
> If noise, etc. interference is expected, you should use a anti-interference extension cable(Shielded or Twisted  Cable or Copper Shield Network etc.).  [(link)](https://www.antclabs.com/wiring)

So if you have a problem like described above, you can easily test it by taking a new wire and lead it "anywhere" but not beside your stepper/heater wires. Test it again if the problem is gone than interference is your problem.

Papooch, 2019-11-18:
Had this same problem happen to me. I rewired everything, even moved the cables away from the motor cables but to no avail.

Having already replaced the BLTouch pin (and tossing the old one away) because I have had this happen before (and changing the pin had fixed it), I decided to inspect it more thoroughly before buying myself a new pin . I discovered that there is a little metal cylinder at the top (that is used to stove the pin using an electromagnet). This little cylinder sits in a plastic tunnel and I noticed that mine was pushed only about half way in. Having nothing else to lose I pushed it all the way down using a toothpick and viola, that did it!

I run the self test and everything and then it stopped working again. And sure enough the little metal cylinder has climbed half way up again, most certainly because of the the vibrations and the continuous up and down clicking of the self test. This time I pushed it down and added a drop of superglue and that fixed it for good (or so I hope).

reloxx13, 2019-12-29:
@Papooch did u reach it through the hole on top if u remove the screw?

my bltouch sudenly stopped working now, too.
printed last night, today it wont move the pin anymore :(

Papooch, 2019-12-29:
@reloxx13 Yes, if you unscrew the grub screw on the top, you can pull the pin out.

reloxx13, 2019-12-29:
Heya,

yes, i did it already and it was fine.

It just flashes and the pin does not move anymore.

I tried to adjust the distance with the screw without success.
I checked all cables and they are fine.
Now i got a fake BLTouch lying around, i connected it with the same wires and it is working fine.

So the BLTouch just died without any reason.