MarlinFirmware/Marlin #12948: Bltouch gets triggered nur z Motor doesnt stop

mpm1396, 2019-01-18:
Hello Guys, i am Using marlin 1.1.8. BL Touch lights up red when triggered but z Motor doesnt stop when homing. All other endstops are working. Printer: anycubic prusa i3. Trigorilla Board arduino 2560 ramps.

mpm1396, 2019-01-18:
config.h `/**
   FOR ANYCUBIC PRUSA I3 WITH BLTOUCH 01/05/2018
   
   Marlin 3D Printer Firmware
   Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]

   Based on Sprinter and grbl.
   Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm

   This program is free software: you can redistribute it and/or modify
   it under the terms of the GNU General Public License as published by
   the Free Software Foundation, either version 3 of the License, or
   (at your option) any later version.

   This program is distributed in the hope that it will be useful,
   but WITHOUT ANY WARRANTY; without even the implied warranty of
   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
   GNU General Public License for more details.

   You should have received a copy of the GNU General Public License
   along with this program.  If not, see <http://www.gnu.org/licenses/>.

*/

/**
   Configuration.h

   Basic settings such as:

   - Type of electronics
   - Type of temperature sensor
   - Printer geometry
   - Endstop configuration
   - LCD controller
   - Extra features

   Advanced settings can be found in Configuration_adv.h

*/
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010107

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
   Here are some standard links for getting your machine calibrated:

   http://reprap.org/wiki/Calibration
   http://youtu.be/wAL9d7FgInk
   http://calculator.josefprusa.cz
   http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
   http://www.thingiverse.com/thing:5573
   https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
   http://www.thingiverse.com/thing:298812
*/

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// example_configurations/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
[... 1687 lines cut ...]
/**
   R/C SERVO support
   Sponsored by TrinityLabs, Reworked by codexmas
*/

/**
   Number of servos

   For some servo-related options NUM_SERVOS will be set automatically.
   Set this manually if there are extra servos needing manual control.
   Leave undefined or set to 0 to entirely disable the servo subsystem.
*/
#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command

// Delay (in milliseconds) before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Servo deactivation
//
// With this option servos are powered only during movement, then turned off to prevent jitter.
//#define DEACTIVATE_SERVOS_AFTER_MOVE

#endif // CONFIGURATION_H`

FanDjango, 2019-01-18:
Has it ever worked before on your printer or is this the initial situation?

FanDjango, 2019-01-18:
Consider reading through the neighboring issue, it contains some tips that might help #12947

boelle, 2019-01-18:
and consider attaching the config's as zip instead of here, makes the post VERY long

mpm1396, 2019-01-19:
well guys I fixed it kinda. The Problem was the polarity of the sensor wires was reversed. Next Problem is when I start the print it starts somewhere on the left side of the bed not in the middle. Auto bed leveling and homing works. When I start pronterface and press the red x y Center button it runs into the  x and y endstop. Here is my config 
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/2775395/conf.zip)

mpm1396, 2019-01-19:
Here is my start gcode maybe it is important
[gcodestart.zip](https://github.com/MarlinFirmware/Marlin/files/2775401/gcodestart.zip)

boelle, 2019-01-19:
you are missing this define

//#define BED_CENTER_AT_0_0

remove //

what version of marlin?

boelle, 2019-01-19:
and please shorten up your first post

mpm1396, 2019-01-19:
Marlin 1.1.8 . First post is already shortened :D

mpm1396, 2019-01-19:
I did defined it, but I am getting an error in Arduino ide "static assertion failed: RIGHT_PROBE_BED_POSITION can't be reached by the Z probe."

boelle, 2019-01-19:
upps i meant 2nd post with the config pasted in :-(

boelle, 2019-01-19:
try #define RIGHT_PROBE_BED_POSITION 110 or a little less

mpm1396, 2019-01-19:
I used 100 now I get 
static assertion failed: BACK_PROBE_BED_POSITION can't be reached by the Z probe.

boelle, 2019-01-19:
ohhh... now i remember things :-D
with #define BED_CENTER_AT_0_0

the center is 0 and one direction is positive number and other direction is negative

so you will most likely also have to adjust these

// The size of the print bed
#define X_BED_SIZE 200
#define Y_BED_SIZE 200

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 160

boelle, 2019-01-19:
if X and Y ednstop is precise at edge of bed i would use:

#define X_MIN_POS -100
#define Y_MIN_POS -100

#define X_MAX_POS 100
#define Y_MAX_POS 100

mpm1396, 2019-01-19:
So should I comment the bed centre 0 0 out again?

boelle, 2019-01-19:
nope, that one stays as is

boelle, 2019-01-19:
i use 0,0 down in the lower left corner and its a long time since i have had a setup with 0,0 in center
and i'm 40 year old so not that fast at remember things

mpm1396, 2019-01-19:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2775414/Configuration.zip)
I did what you said and still gettin error: static assertion failed: RIGHT_PROBE_BED_POSITION can't be reached by the Z probe.

boelle, 2019-01-19:
just me old

#define RIGHT_PROBE_BED_POSITION 110

needs to be less than 100, but i cant remember how much lees

#define BACK_PROBE_BED_POSITION 190

also needs to be less than 100

#define FRONT_PROBE_BED_POSITION 20

can be closer to 100

remember than some of the 100's need to be negative 100

boelle, 2019-01-19:
but when you home the printer now it will be very clear which of the 100's that need to be negative

mpm1396, 2019-01-19:
When I home the Printer it goes to x 25 y 93 z 16.7

boelle, 2019-01-19:
the reason i dont know which of the 100's that need to be negative is that i dont know if right side is the side it homes to

and if front or back is where Y homes

boelle, 2019-01-19:
x 25 y 93 z 16.7

that makes sense

#define X_PROBE_OFFSET_FROM_EXTRUDER 75  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 7  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.7   // Z offset: -below +above  [the nozzle]

25+75 = 100

93 + 7 = 100

i cant remember the logic on Z

boelle, 2019-01-19:
how do you home?
 G29?

mpm1396, 2019-01-19:
with the screen. prepare -- > autohome

boelle, 2019-01-19:
its like it homes the probe

try send G28 and see if that makes any difference

mpm1396, 2019-01-19:
no difference with G28

boelle, 2019-01-19:
and after G28

try send G0 X0 Y0

boelle, 2019-01-19:
what we want to figure out  is where printer thinks it is so we can change config and get it to be where we want it

how close to center was it after G28? and did it go to endstop first

mpm1396, 2019-01-19:
so with g28 it homes somewher on the left side. If I send G0 X0 Y0 it runs into the Y endstop.

boelle, 2019-01-19:
with G28 it should touch the endstops

do it touch the endstops?

boelle, 2019-01-19:
G28 X if you only want to try and home X

mpm1396, 2019-01-19:
Yes with g28

boelle, 2019-01-19:
oki.... and it stays at the endstop after G28 ?

boelle, 2019-01-19:
what is strange is that with G28 it should home and be at -100 for X and Y

Try send G28 and then M114 and tell me what comes on serial

i'm looking at your latest config to see if we made a mistake

mpm1396, 2019-01-19:
Its this one http://m.anycubic3d.com/support/show/594036.html
Its the prusa i3 Kit with Ultrabase.

mpm1396, 2019-01-19:
The config ist from Here. https://www.thingiverse.com/groups/anycubic-prusa-i3/forums/general/topic:26690

boelle, 2019-01-19:
config looks ok

Try send G28 and then M114 and tell me what comes on serial

boelle, 2019-01-19:
and it just came to mind that after homing it might move some MM away from the endstop

so if you send G28 and then M114 at it says X98 (or close to that) then our config is ok

mpm1396, 2019-01-19:
so now i am back. I did g28 and M114. this is the Output X:25.00 Y:93.00 Z:16.70 E:0.00 Count X:2000 Y:7440 Z:6680

boelle, 2019-01-19:
that is strange... can you attach configuration.h again as zip

and make sure its not an old copy :-D i have made that mistake many times

mpm1396, 2019-01-19:
I just reflashed this: 
[Configurationf.zip](https://github.com/MarlinFirmware/Marlin/files/2775501/Configurationf.zip)
this is still the Output of M114
X:25.00 Y:93.00 Z:16.70 E:0.00 Count X:2000 Y:7440 Z:6680

boelle, 2019-01-19:
you did not change these:

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 160

boelle, 2019-01-19:
you did not change this one either

//#define BED_CENTER_AT_0_0

boelle, 2019-01-19:
but that still does not explain why X is at 25 after home

@thinkyhead can just shine the light of god wisdom on this one?

mpm1396, 2019-01-19:
I uncommented and changed x_min_pos to -100 y min -100 x max 100 y max 100 but I get the error: 
static assertion failed: RIGHT_PROBE_BED_POSITION can't be reached by the Z probe.

mpm1396, 2019-01-19:
Well i switched front and back now i get static assertion failed: LEFT_PROBE_BED_POSITION can't be reached by the Z probe.

boelle, 2019-01-19:
dont change it back... need to think

boelle, 2019-01-19:
what side of nozzle is probe? right side?

boelle, 2019-01-19:
and 75 mm from nozzle right?

mpm1396, 2019-01-19:
ok I got it like this at the Moment:
#define LEFT_PROBE_BED_POSITION -100
#define RIGHT_PROBE_BED_POSITION 100
#define FRONT_PROBE_BED_POSITION -100
#define BACK_PROBE_BED_POSITION 100

#define X_MIN_POS -100
#define Y_MIN_POS -100
#define Z_MIN_POS 0
#define X_MAX_POS 100
#define Y_MAX_POS 100
#define Z_MAX_POS 160

#define BED_CENTER_AT_0_0

boelle, 2019-01-19:
see me 2 comments above yours :-D

mpm1396, 2019-01-19:
so like this? :D

#define FRONT_PROBE_BED_POSITION 100
#define BACK_PROBE_BED_POSITION -100

boelle, 2019-01-19:
what side of nozzle is probe? right side?
and 75 mm from nozzle right?

and no dont change anything

mpm1396, 2019-01-19:
my bltouch is around 7cm on the Right side from the nozzle

mpm1396, 2019-01-19:
like in this Picture here https://www.thingiverse.com/thing:2887389

boelle, 2019-01-19:
oki... try:

#define LEFT_PROBE_BED_POSITION -25

explanation: since probe is 75mm on right side of nozzle it can never get to -100, the best it can is -25

boelle, 2019-01-19:
do you get an error for #define FRONT_PROBE_BED_POSITION ??

mpm1396, 2019-01-19:
I get a lot of Errors in Arduino IDE with this values: 
#define LEFT_PROBE_BED_POSITION -25
#define RIGHT_PROBE_BED_POSITION 100
#define FRONT_PROBE_BED_POSITION -100
#define BACK_PROBE_BED_POSITION 100
#define X_MIN_POS -100
#define Y_MIN_POS -100
#define Z_MIN_POS 0
#define X_MAX_POS 100
#define Y_MAX_POS 100
#define Z_MAX_POS 160

[555 more replies]

closed 2019-01-21 by mpm1396