MarlinFirmware/Marlin #12952: [bugfix-2.0.x] Printer Halted - Auto Home (G28) from position 0.00

Adinfo78, 2019-01-19:
Printer: Anycubic Linear Plus
Firmware vesion: Marlin 2.0.x bugfix

Steps:
1. Set Delta Height
2. Manually move head to 0.00 (the sheet of paper passes with resistance)
2. Auto Home
3. Printer Halted

Pronterface console:
SENDING: G28
echo: busy: processing
Error: Printer halted. kill () called!
[ERROR] Error: Printer halted. kill () called!

Details of the problem described:
1. Delta Height set in such a way that at 0.00 a paper sheet passes with resistance.
2. At this point, want to run the Auto Home (G28) carts go up, but at the end there is a jam - Printer Halted, Please Restart. The carts are then about 1 cm under the limits.
3. If the head is higher, eg 5cm above the bed, then Auto Home is working properly.
4. The problem occurs both in bugfix version 1.1.9 and bugfix version 2.0.
5. In version 1.1.9 without bugfix, Auto Home works fine, except for the problem described here (https://github.com/MarlinFirmware/Marlin/issues/11676).
6. When setting the probe height to 15.08 after calibration, the value for delta height is ideal.
7. Problem with Auo Home in Marlin 2.0 bugfix in the same printer confirmed by a friend.
8. Colleagues from the reprapy.pl forum suggested that the Z_MAX_POS variable may be a problem, but I compared the configurations of different MARLIN versions and is the same everywhere.
9. I noticed a difference compared to version 1.1.9 (without bugfix) - there after Delta Height calibration indicates a different value:
Marlin 1.1.9 = 294.68
Marlin 1.1.9 and 2.0 bugfix = 278.92
Difference: 15.76 (approximately).
10. The same happens if I cancel printing on the first few layers and run Auto Home.

I tried to eliminate the interference in the limit switches, I added some elements and connected the pullup
(a 100 ohm resistor in series with a terminal, a 100nF capacitor in parallel, a 10kOhm resistor together with a diode for this 5V vcc) - this modification did not bring any improvement in the presented problem.

In any case, the difference between these versions results from something, the carriages on the guides going up to Auto Home get a limit where they stop for safety reasons. Sometimes a single limit in this situation becomes compact, but the other two do not, however, in most cases, all the trolleys stop minimally under the limits of the limit switches.

My configuration: http://retroroom.pl/reprapy/Marlin_Anycubic_Delta_Linera_Plus_Adinfo78.zip

jmdearras, 2019-01-19:
I can confirm this.  Trying to move to 2.0.x, I get exactly the same error from the same situation.

jmdearras, 2019-01-20:
OK, prelim, it seems to be related to probe height.  I had changed mine to 15.1 (my paper drag level) and I get the same failure on 1.1.9.  makes no sense but I am testing, now

KA8888, 2019-01-20:
Same problem here, and even bugfix 1.1.9 has this bug.
I already wrote an issue on this a couple days ago:
https://github.com/MarlinFirmware/Marlin/issues/12880

shitcreek, 2019-01-23:
Looks like this issue is resolved, just awaiting the pull request to be accepted. In the meantime, please consider closing out the issue. Thanks.

Adinfo78, 2019-01-23:
The problem with delta homing has been resolved after correction in the file Marlin/src/module/delta.cpp ([bugfix-2.0.x] Fix the delta homing issue #12974).

closed 2019-01-23 by Adinfo78