MarlinFirmware/Marlin #12954: [2.0.x]TMC2208 UART won't start in stealthchop

jvdbrg, 2019-01-19:
The TMC2208 UART connected drivers won't start in stealtchop.

1. Upload firmware
2. Do a G28

Because I configured the drivers to operate in stealthchop until a certain threshold I would expect the drivers to run in stealtchop and not in spreadcycle.

Dre drivers however run in spreadcycle and make a terrible noise.
They will run in Stealtchop by setting them in the LCD menu.

However when I store the parameters this setting doesn't seem to be stored.

Strange thing, When I do a M503 the drivers are imidiately activated and are in spreadcycle.

teemuatlut, 2019-01-19:
How does the printer sound if you do a `G1 X10` right after the reboot when the driver is supposed to be in stealthChop but is in spreadCycle instead?

Can you give the `M122 V` output before and after movement? You'll need `TMC_DEBUG` for this.

It'll help if you can give the debugging information at every step of the way.

teemuatlut, 2019-01-19:
I can see from your GCONF output that the driver truly is in spreadCycle mode both before and after the movement. I'm guessing I missed something as I did most of my testing with TMC2130 where the logic is flipped compared to TMC2208.
Thanks for letting me know and I'll be able to look into this more some time next week.

> this works great BTW even when the printer is running :)

Good to hear!

boelle, 2019-02-21:
@jvdbrg problem solved?

closed 2019-03-03 by thinkyhead