MarlinFirmware/Marlin #12957: [2.0] FR to improve advanced pause / filament change

hamster65, 2019-01-19:
- When my printer parks after the filament runout sensor has been triggered, the layer fan stays on. This is not a major thing, but it could be improved by switching it off during pause.

- What would be more important: Prime the nozze when leaving the pause mode.
Right now, the filament gets loaded and purged. After that, filament oozes out of the nozzle, so when you hit the button to continue, filament is missing for a couple of mm.

InsanityAutomation, 2019-01-21:
2 and 3 have been things I was considering doing at some point soon. For 2, I was looking at an extra restart distance to bring in after returning on tool change, and it can be utilized in the same way on return from pause.

reloxx13, 2019-03-08:
i used the nozzle park and change filament feature today and its seems to be buggy.

the main porpose works, but...

filament was empty when i was not at home, so i shutted it down remotly.
came home and tried the unlaod function, but the extruder could not reach the filament anymore, nevermind.

- no homing before nozzle park, had to power off the machine before it hits the top x bar.
-> it should home or ask for home or dont go to nozzle park feature if not homed.

- texts (german) were completly wrong while loading. it showed "unload more" all over instead of load more (headline, help text).
showed "continue print", did not print, started it right after powering on.
-> expected the right text

- after loading, heating was not disabled.
-> i was not printing, so i expected that the heating stops after unload/load. it was not heating before i hitted load. i could choose the pre-configured pre-heat menu entry.

overall it worked, but this moving up to park position w/o home should be prevented to avoid damage!
and heating stayed on could be dangerous if a user does not pay attention and disabled heating manual.

still i ove this feature an would love it even more if things get fixed <3
i hope this issue wont get lost :)

boelle, 2019-06-22:
wrong label

InsanityAutomation, 2019-06-22:
Actually this can be closed. Its done.

boelle, 2019-07-21:
i trust @InsanityAutomation on this one and will close it assuming that PR##12984 did the trick

closed 2019-07-21 by boelle