MarlinFirmware/Marlin #12958: [FR] MK3 runout sensor

GMagician, 2019-01-19:
Soon (maybe a week) I'll get a sensor and I'm going to implement it...that's a big challenge to me but how knows...

I'm here to ask to "Best marlin guys" what it's better approach.
I don't know how MK3 firmware does it but for sure it's far from current Marlin runout implementation.
I may try do follow 2 paths:
1) include MK3 code as indepented runut logic and adapt it to Marlin;
2) include MK3 adapting it to current Marlin implemented logic (I'm thinking about something like encoder sensor)

MK3 sensor uses i2c bus (SDA & SCL) so I think an interrupt to handle communication but despite of this everything else may safely follow 'FilamentSensorEncoder' like class logic

What do you think?

dcwalmsley, 2019-01-19:
@Giuliano Zaro,

I recently incorporated Prusa's MMU2 upgrade kit to my HyperCube Core XY 
printer and with the help of @Revilor and @Ruud Rademaker's information I 
have the ability to reset MMU, and properly feed my specific PTFE tubing 
(565mm) for loading to and extracting from the direct-drive extruder.  What 
I need now is what you wrote, code to enable the optical sensor using 
(5vdc, Gnd, SDA, and SCL) to sense the filament for filament runout (M600).

I would be grateful if Devs could look into code to enable this feature or 
a work-around solution.

Best Regards,
Doug Walmsley

GMagician, 2019-01-20:
Move discussion to PR

closed 2019-01-20 by GMagician