MarlinFirmware/Marlin #12960: Cannot understand the concept of DEFAULT_ACCELERATION...

Louis-Cypher, 2019-01-19:
### Description
I am having issues with my Z- Axis and I guess it is related to the introduction of DEFAULT_ACCELERATION. Hence I do want to understand the concept of this part of the Configuration.h

```
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }

/**
 * Default Acceleration (change/s) change = mm/s
 * Override with M204
 *
 *   M204 P    Acceleration
 *   M204 R    Retract Acceleration
 *   M204 T    Travel Acceleration
 */
#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   3000    // X, Y, Z acceleration for travel (non printing) moves

``` 

1. are those three values in the last 3 lines limited to what is defined in DEFAULT_MAX_ACCELERATION?
2. What is the concept having dedicated X,Y,Z values for DEFAULT_MAX_ACCELERATION but not in the DEFAULT_XXX_ACCELERATION values. I do not understand why there is no distinction between axes here
3. Just for the sake of curiosity: Why aren't the acceleration values related to steps rather to mm? Assuming that most printers use identical motors for at least X,Y, and Z this would make tuning accelerations much easier (if the optimal accelerations are motor related rather than related to the mechanical properties of the frame).
 
**Expected behavior:** 
Have a default acceleration value that is axis dependent

**Actual behavior:** 
One value for all axes

forkoz, 2019-01-19:
The max are all limits. The defaults are what the printer uses. So if your Z is too fast, limit the max of the Z.. also maybe the feed rate too. The purpose of combined values is so you can tune them on the fly.

The limits of your motion will be mechanical and very unlikely motor related.

Louis-Cypher, 2019-01-19:
Thanks for clarification. That is what I assumed. But I do not understand why there are "generic" defaults , since they in fact never will apply to e.g. the Z- Axis since this axis has an order of magnitude lower max values. To me this appears kind of inconsistent. If there is a specific max value for every axis from my understanding there should be a specific default value for every axis as well.

Louis-Cypher, 2019-01-20:
> Whatever values you may have set later for your max accelerations,
> executing M502 will restore them back to the values set in
> DEFAULT_MAX_ACCELERATION, as with other default settings.

Got that. Maybe I should give an example of what I mean:

MAX_DEFAULT_ACCELERATION is 3000, 3000, 100, 500

DEFAULT_ACCELERATION is  2000

That does mean that DEFAULT_ACCELERATION will *never* be applied to the z- 
axis since its MAX_DEFAULT_ACCELERATION is an order of magnitude lower than 
DEFAULT_ACCELERATION. If I lower it to a range applicable for z, the values 
will be an order of magnitude too low for x and y. Ergo those three default 
values in fact are not usable for the z- axis, right?

swilkens, 2019-01-20:
DEFAULT_ACCELERATION is used by default and is limited by DEFAULT_MAX_ACCELERATION. In your case, the printer will use:

* X 2000 (since this is default)
* Y 2000 (since this is default)
* Z 100 (limited by DEFAULT_MAX_ACCELERATION)
* E 500 (limited by DEFAULT_MAX_ACCEERATION)

You can use GCODE to instruct the printer to set a new DEFAULT_ACCELEARTION during a print to modify the acceleration for a section of the print. That, too, will be limited by DEFAUT_MAX_ACCELERATION.

Note that the DEFAULT_MAX_ACCELERATION is also defined on an axis level. It is NOT compounded. (i.e. no print = sqrt ( x^2 + y^2 + z^2 ).

Louis-Cypher, 2019-01-20:
Completely understood (from the beginning). I just didn't get why Z in fact doesn't have a DEFAULT_ACCELERATION by design of the software whereas, X and Y do. Since we seem to share how it works and developers obviously have understood what I think it should be, but don't share my opinion that this is kind of inconsequent I think we can close this issue.

closed 2019-01-20 by Louis-Cypher

swilkens, 2019-01-20:
It is mildly confusing that the default definition differs from the max definition, I agree. I asked this exact question a few months back, as I was equally confused.