MarlinFirmware/Marlin #12961: Feature request: support >3 fans in Marlin 2.0

Bougakov, 2019-01-19:
This is a feature request. **Marlin 2.0 bugfix** source provides support for up to 5 extruders but only for up to 3 fans.

Additional "fan" outputs might be used to drive extra equipment: in my case it is a secondary extruder head assembly that carries a 405nm laser and micro drill (for making PCBs using resist film from clean copper boards).

[image]

Unfortunately, adding such support requires the user to make significant amount of changes in multiple files which takes a lot of effort. 

I suggest to support equal number of fans and extruders - five. This will allow users to implement advanced confiurations while the "laser overhaul" (https://github.com/MarlinFirmware/Marlin/issues/11576#issuecomment-436501316) is not finished. In my case the following modifications were needed:

In board's pin definition file:

```
#define E0_AUTO_FAN_PIN     9
#define FAN1_PIN            8
#define FAN2_PIN            5 // Previously left "servo" pin, currently laser
#define FAN3_PIN            3 // Previously right "servo" pin, currently a micro drill
```

in `pins.h`

```
#ifndef FAN_PIN
  #define FAN_PIN -1
#endif
#ifndef FAN1_PIN
  #define FAN1_PIN -1
#endif
#ifndef FAN2_PIN
  #define FAN2_PIN -1
#endif
#ifndef FAN3_PIN // More fans
  #define FAN3_PIN -1
#endif
#ifndef FAN4_PIN // More fans
  #define FAN4_PIN -1
#endif
#ifndef FAN5_PIN // More fans
  #define FAN5_PIN -1
#endif
```

in `Conditionals-post.h`:

```
/**
 * Up to 3 PWM fans
 */
#if HAS_FAN5 // More fans
  #define FAN_COUNT 6 
#elif HAS_FAN4 // More fans
  #define FAN_COUNT 5 
#elif HAS_FAN3 // More fans
  #define FAN_COUNT 4 
#elif HAS_FAN2
  #define FAN_COUNT 3
#elif HAS_FAN1
  #define FAN_COUNT 2
#elif HAS_FAN0
  #define FAN_COUNT 1
#else
  #define FAN_COUNT 0
#endif

#if HAS_FAN0
  #define WRITE_FAN(v) WRITE(FAN_PIN, v)
  #define WRITE_FAN0(v) WRITE_FAN(v)
#endif
#if HAS_FAN1
  #define WRITE_FAN1(v) WRITE(FAN1_PIN, v)
[... 236 lines cut ...]
        MENU_MULTIPLIER_ITEM_EDIT(int8, MSG_EXTRA_FAN_SPEED " 6", &new_fan_speed[5], 3, 255);
      #endif
    #endif
    #endif // FAN_COUNT > 0
```

and, finally, in `sensitive_pins.h`:

```
#if PIN_EXISTS(FAN3) // More fans
  #define _FAN3 FAN3_PIN,
#else
  #define _FAN3
#endif
#if PIN_EXISTS(FAN4) // More fans
  #define _FAN4 FAN4_PIN,
#else
  #define _FAN4
#endif
#if PIN_EXISTS(FAN5) // More fans
  #define _FAN5 FAN5_PIN,
#else
  #define _FAN5
#endif
```

boelle, 2019-01-23:
i could use this (i think)

i only have 1 extruder but i need to control 1 more fan than the default pins file for the re-arm has

boelle, 2019-03-13:
@Bougakov still working on this?

ogland, 2020-07-29:
Not sure why this isn't a thing yet. Please make it real.

ellensp, 2020-07-29:
fans 0 through fan 7 have been on marlin for some time...

ogland, 2020-07-29:
Yeah! I just NOW found that out when attempting the same method I did a while back using this: https://github.com/MarlinFirmware/Marlin/issues/12961

Turns out it was really straight forward. Weird that the issue isn't closed.

ellensp, 2020-07-29:
I can fix that :)

closed 2020-07-29 by ellensp