MarlinFirmware/Marlin #12963: [2.0] TMX2130 Issue on X-axis with GT2560.

pulster, 2019-01-20:
Hello!

I've added a few weeks ago a TMC2130 on my Y-axis. It worked well.
Now I added a TMC2130 on my X-axis and it doesn't work. Y axis works well.

Here is an output of M122:

Send: M122
Recv: 		                  X	        Y
Recv: Enabled		false	false
Recv: Set current	         800	800
Recv: RMS current	   795	795
Recv: MAX current	  1121	1121
Recv: Run current	        25/31	25/31
Recv: Hold current	12/31	12/31
Recv: CS actual		31/31	12/31
Recv: PWM scale	          255	0
Recv: vsense		       1=.18	1=.18
Recv: stealthChop	         true	true
Recv: msteps		              0	16
Recv: tstep	 4294967295	max
Recv: pwm
Recv: threshold		      0	0
Recv: [mm/s]		              -	-
Recv: OT prewarn	         true	false
Recv: off time		     15	3
Recv: blank time	           54	24
Recv: hysteresis
Recv: -end		            12	-1
Recv: -start		              8	1
Recv: Stallguard thrs	      0	0
Recv: DRVSTATUS	      X	 Y
Recv: stallguard		                X
Recv: sg_result		 1023	0
Recv: fsactive		X
Recv: stst		        X	      X
Recv: olb		        X
Recv: ola		                X
Recv: s2gb		        X
Recv: s2ga		        X
Recv: otpw		         X
Recv: ot		                 X
Recv: Driver registers:
Recv: 		X	0xFF:FF:FF:FF	 Bad response!
Recv: 		Y	0x80:0C:00:00
Recv: Testing X connection... Error: All HIGH
Recv: Testing Y connection... OK

Does anyone know how I can get them to work?
I've replaced reversed X and Y drivers, still X doesn't work and Y does work.

Thank you.

teemuatlut, 2019-01-20:
Bad comms -> check wiring. Preferably with a multimeter.

Read http://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting

pulster, 2019-01-23:
So I triple checked everything and it was all OK.
I changed my pins to E1 and moved the driver. Everything works.

Still don't know what exactly was the problem. Maybe a bad pin on the connectors.

Anyways, now it works!

closed 2019-01-23 by pulster