MarlinFirmware/Marlin #12964: MKS-SBase huge moire

J-PN, 2019-01-20:
My set-up is X5S (coreXY) with SBase v1.3 (+ 8 diode smothers but didn't do much) & Marlin2.0 17Jan19

Seems that layer shift is OK with 
#define MINIMUM_STEPPER_PULSE 4

but what I cannot fix is the bad moire pattern
[image]
The test part is D-shape egg. The curve section is smooth but Y-direction strait part has the pattern. Seems that X&Y motors aren't rotating equal speeds but oscillates. (bottom pattern is 'cause Ultrabase) 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2776468/Configuration.zip)
----
The secondary issue came up as well, used to work in past.
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
compiling error:

_Compiling .pioenvs\LPC1768\src\src\module\printcounter.cpp.o
In file included from Marlin\src\module\../inc/MarlinConfig.h:49:0,
from Marlin\src\module\configuration_store.h:24,
Compiling .pioenvs\LPC1768\src\src\module\probe.cpp.o
from Marlin\src\module\configuration_store.cpp:47:

Marlin\src\module\configuration_store.cpp: In function 'void say_M603()':
Marlin\src\module\configuration_store.cpp:2346:52: error: 'port' was not declared in this scope
void say_M603(PORTARG_SOLO) { SERIAL_ECHOPGM_P(port, "  M603 "); }_
------
if I comment the //#define SERIAL_PORT_2 -1
compiling goes OK but obviously MKS TFT display won't work.

gloomyandy, 2019-01-20:
You should probably create a separate issue for the serial port compilation error, otherwise it may get missed.

As to your other problem, what makes you think that the moire problem is a Marlin issue? Did an earlier version not show the problem? Does some other firmware print the same gcode without a problem?

forkoz, 2019-01-20:
Heh, I get the moire/ghosting too. Smoothers didn't do much in that regard. I use S_curve, Junction_d, etc. Just kind of live with it. Since I have a cartesian and different mechanics I'm guessing its related to the DRV8255 and not anything mechanical. Even upgraded Y bearings and X to linear rail. Still ghost and have rounded corners.

kAdonis, 2019-01-20:
I also had this moire before with DRV8825 on my RAMPS/ Re-Arm, it was getting better after I cranked up the extruder motor current, but it was still visible
I changed the extruder driver for a A4988 and the moire was gone 
with SERIAL_PORT 0 defined, the TFT should work
with `//#define SERIAL_PORT_2 -1` the USB does not work
I can compile fine for MKS SBase with:

```
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
```

RudyBenner, 2019-01-20:
Your line 991 gives me several errors in compile. When I comment it out, the compile sails through.

J-PN, 2019-01-20:
> You should probably create a separate issue for the serial port compilation error, otherwise it may get missed.
>
> As to your other problem, what makes you think that the moire problem is a Marlin [...]

With the printer's original card, same gcode, (Marlin 1.1.9) the surface is totally smooth, so this is not mechanical issue.

gloomyandy, 2019-01-20:
Did your original card have the same drivers (DRV8825)? Have you tried running smoothieware on your new card and printing the same gcode?

gloomyandy, 2019-01-21:
If you can it would be worth trying smoothieware to see if the problem is there with that (I suspect it will be). 

Oh and what voltage are you running the steppers at? My understanding is that those diode smoothers do not work as well when using 24V as with12V (though other things may be better).

J-PN, 2019-01-21:
I had to install the original board to get some work done but probably next weekend I have time to swap the SBase back.

On the curved areas the salmon skin is visible but with the smoothers the artifact is smoother, though on my eye more disturbing waves. PSU is 11.58V

J-PN, 2019-01-29:
Today I tried the 27-Jan-19 build and the outcome is totally smooth. I re-installed the Atom updated the conf-files. 

Still having the issue about the 2nd serial port, need to make the other item.

closed 2019-01-29 by J-PN