MarlinFirmware/Marlin #12968: G29 H - Asking just for a check

MasterPIC, 2019-01-20:
Hi,

G29 H (please refer directly to #12242 and #12610) probing works as expected.
But **I'm not sure** left and front (starting) probe positions and grid size (in mm) are stored/restored by M500/M501.
I found no commands to ask the printer to report starting probe positions and grid size. So I'm not sure the grid, after being stored to EEPROM, is still valid (within the area it should cover and limited to) if the printer will be switched off.

I'd appreciate a confirmation.

Thanks
Armando

RudyBenner, 2019-01-20:
M115 will spit out some of what you need. You get to decode it yourself.

MasterPIC, 2019-01-21:
I think M115 doesn't report the required info.
I just opened a new feature request: #12979

MasterPIC, 2019-01-21:
In the meantime I found by myself that
bilinear_start[X_AXIS]
bilinear_start[Y_AXIS]
bilinear_grid_spacing[X_AXIS]
bilinear_grid_spacing[Y_AXIS]
are properly stored to (and restored from) EEPROM.

closed 2019-01-21 by MasterPIC