MarlinFirmware/Marlin #12970: Anycubic I3 homes fine but crash's Y endstop on any move with X and Y

mpm1396, 2019-01-20:
original issue: https://github.com/MarlinFirmware/Marlin/issues/12948
checked list:
endstops are connected correct
X and Y homes ok
when sending G28 X followed by G0 X10 nothing happens
when sending G28 Y followed by G0 Y10 it crashes in to Y endstop
checked that both X and Y steppers are not blown
when sending G28 Z followed by G0 Z10 Z does not move and bed crash in to endstop
when homing and moving from the lcd everything works fine

kAdonis, 2019-01-20:
@mpm1396 is using a Trigorilla board, this board has its own board definition in marlin
and the E1 driver pins get undefined, so they are not usable in marlin
you should probably use only one z driver for both z motors

boelle, 2019-01-20:
we moved to marlin 2.0 and things are working

only the heatsink/hotend fan will not come on at 50C

configuration_adv_.h: #define E0_AUTO_FAN_PIN TRIGORILLA_FAN2_PIN
we also tried with: #define E0_AUTO_FAN_PIN 44